zwave-js/zwave-js #6564: Unknown Manufacturer/Product/Product Code after upgrade

renatohtpc, 2023-12-20:
After upgrading from version 9.3.2 to 9.5 (and later 9.6) I noticed that several of my devices are now showing as "Unknow" as depicted from the attached.  I was able to fix a few by removing them and re-adding them.  

[image]

After. Trying to reinterview the affected devices did not help.

I am running the software inside a docker. Here is the script:

`root@IOT:~/docker_compose/zwavejs2mqtt# more docker-compose.yml
version: '3.7'
services:
  zwavejs2mqtt:
    container_name: zwavejs2mqtt
    image: zwavejs/zwavejs2mqtt:latest
    restart: always
    tty: true
    stop_signal: SIGINT
    environment:
      - SESSION_SECRET=XXXXXXXXXXXXXXXX
      - ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db
      # Uncomment if you want logs time and dates to match your timezone instead of UTC
      # Available at https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
      - TZ=America/New_York
    networks:
      - zwave
    devices:
      # Do not use /dev/ttyUSBX serial devices, as those mappings can change over time.
      # Instead, use the /dev/serial/by-id/X serial device for your Z-Wave stick.
      - '/dev/serial/by-id/usb-Zooz_800_Z-Wave_Stick_533D004242-if00:/dev/zwave'
    volumes:
      #- ./zwave-config:/usr/src/app/store
      - $PWD/store:/usr/src/app/store
    ports:
      - '8091:8091' # port for web interface
      - '3005:3000' # port for Z-Wave JS websocket server
networks:
  zwave:
#volumes:
#  zwave-config:
#    name: zwave-config`

Also, here is the output of the 'ls' command for the USB stick:

`root@IOT:~/docker_compose/zwavejs2mqtt# ls -al /dev/serial/by-id/usb-Zooz_800_Z-Wave_Stick_533D004242-if00
lrwxrwxrwx 1 root root 13 Dec  2 21:36 /dev/serial/by-id/usb-Zooz_800_Z-Wave_Stick_533D004242-if00 -> ../../ttyACM0`

I am also attaching the log file (large)

[zwavejs_2023-12-19.log](https://github.com/zwave-js/zwave-js-ui/files/13720064/zwavejs_2023-12-19.log)

Is removing/adding the only solution?

Thank you for your help.

Renato

robertsLando, 2023-12-20:
I see a ton of:

```
2023-12-20T05:05:59.664Z DRIVER « [Node 029] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -83 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: No security class granted
```

Did you changed your security keys for some reason or maybe them are not set?

renatohtpc, 2023-12-20:
As far as I can remember, I did not make any changes to them.  Here is a screenshot.

[image]

robertsLando, 2023-12-20:
@renatohtpc them are private, I suggest you to don't upload them

robertsLando, 2023-12-20:
Wait a moment, maybe @AlCalzone can help you. This problem should be debugged on driver side, I moved the issue to his repo but he is actually on parental leave so he may be slow to respond

AlCalzone, 2024-10-08:
Sorry, this slipped through. Is it still an issue?

closed 2024-10-26 by zwave-js-assistant[bot]