zwave-js/zwave-js #6584: Missing device configuration: Building 36 Technologies ADC-SWM150 (ValueError)

Anto79-ops, 2024-01-02:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Building36 Smart Water Valve & Meter ADC-SWM150

### Manufacturer ID

0x0190

### Product Type

0x0007-0x0001

### Product ID

0x0001

### Firmware Version

1.30

### Do you have a manual?

problem is when temperature of water goes below 5oC a temperature alarm is given but its not handled correctly in z-wave js

```
Logger: homeassistant
Source: components/sensor/__init__.py:623
First occurred: 12:16:58 PM (1 occurrences)
Last logged: 12:16:58 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1278, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1062, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 999, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 947, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 623, in state
    raise ValueError(
ValueError: Sensor sensor.smart_water_valve_meter_water_temperature_alarm_status provides state value '8', which is not in the list of options provided
```
```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:507
Integration: Sensor (documentation, issues)
First occurred: 12:16:58 PM (1 occurrences)
Last logged: 12:16:58 PM

Error adding entities for domain sensor with platform zwave_js
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1278, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1062, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 999, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 947, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 623, in state
    raise ValueError(
ValueError: Sensor sensor.smart_water_valve_meter_water_temperature_alarm_status provides state value '8', which is not in the list of options provided
```
```
Logger: homeassistant.components.zwave_js
Source: components/zwave_js/__init__.py:891
Integration: Z-Wave (documentation, issues)
First occurred: 12:16:50 PM (1 occurrences)
Last logged: 12:16:50 PM

Unexpected exception: Sensor sensor.smart_water_valve_meter_water_temperature_alarm_status provides state value '8', which is not in the list of options provided
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/__init__.py", line 891, in client_listen
    await client.listen(driver_ready)
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/client.py", line 284, in listen
    await self.receive_until_closed()
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/client.py", line 443, in receive_until_closed
    self._handle_incoming_message(data)
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/client.py", line 532, in _handle_incoming_message
    self.driver.receive_event(event)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/model/driver.py", line 86, in receive_event
    self.controller.receive_event(event)
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/model/controller/__init__.py", line 854, in receive_event
    node.receive_event(event)
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/model/node/__init__.py", line 462, in receive_event
    self.emit(event.type, event.data)
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/event.py", line 65, in emit
    listener(data)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/entity.py", line 250, in _value_changed
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1062, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 999, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 947, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 623, in state
    raise ValueError(
ValueError: Sensor sensor.smart_water_valve_meter_water_temperature_alarm_status provides state value '8', which is not in the list of options provided
```

apella12, 2024-01-09:
This could be an issue with the home assistant configuration.  Does the sensor have an option for "8" in the .py for water?

The device could be sending "8" as it is a valid Zwave notification.
```
	"0x05": {
		"name": "Water Alarm",
		"variables": [
			{
				"name": "Sensor status",
				"states": {
					"0x01": {
						"label": "Water leak detected (location provided)",
						"description": "The event parameters contain a location report"
					},
					"0x02": {
						"label": "Water leak detected"
					}
				}
			},
			{
				"name": "Maintenance status",
				"states": {
					"0x05": {
						"label": "Replace water filter"
					}
				}
			},
			{
				"name": "Water flow alarm status",
				"states": {
					"0x06": {
						"label": "Water flow alarm",
						"params": {
							"type": "enum",
							"values": {
								"0x01": "No data",
								"0x02": "Below low threshold",
								"0x03": "Above high threshold",
								"0x04": "Max"
							}
						}
					}
				}
			},
			{
				"name": "Water pressure alarm status",
				"states": {
					"0x07": {
						"label": "Water pressure alarm",
						"params": {
							"type": "enum",
							"values": {
								"0x01": "No data",
								"0x02": "Below low threshold",
								"0x03": "Above high threshold",
								"0x04": "Max"
							}
						}
					}
				}
			},
			{
				"name": "Water temperature alarm status",
				"states": {
					"0x08": {
						"label": "Water temperature alarm",
						"params": {
							"type": "enum",
							"values": {
								"0x01": "No data",
								"0x02": "Below low threshold",
								"0x03": "Above high threshold"
							}
						}
					}
				}
			},
```

Anto79-ops, 2024-01-09:
Hi! sorry, I was not sure where to submit this issue, and was going to submit in core.

I don't know how to answer your question, might be able to provide driver logs if I can catch it, if you like?

Anto79-ops, 2024-01-09:
Here it is: 

[zwave_js-5ec3e01351396fb32e419be34214b2dc-Smart Water Valve & Meter-2bcdc1ea3359c1a74882ea3671db67b1.json.txt](https://github.com/zwave-js/node-zwave-js/files/13880982/zwave_js-5ec3e01351396fb32e419be34214b2dc-Smart.Water.Valve.Meter-2bcdc1ea3359c1a74882ea3671db67b1.json.txt)

Anto79-ops, 2024-01-10:
ok, I was able to catch it in the logs, I ran some water deviced LED flashed "5 times" indicated a water temp alarm.

Core gave this error.

```
Logger: zwave_js_server
Source: /usr/local/lib/python3.11/site-packages/zwave_js_server/event.py:68
First occurred: 6:36:49 PM (1 occurrences)
Last logged: 6:36:49 PM

Error handling event: value updated
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zwave_js_server/event.py", line 66, in emit
    listener(data)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/entity.py", line 250, in _value_changed
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1062, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 999, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 947, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 618, in state
    raise ValueError(
ValueError: Sensor sensor.smart_water_valve_meter_water_temperature_alarm_status provides state value '8', which is not in the list of options provided
```

and driver logs gave this when it was in alarm state:

```
2024-01-10T01:36:49.410Z DRIVER « [Node 057] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -64 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 123
                                    │ security class:  S2_Authenticated
                                    └─[SupervisionCCReport]
                                        session id:          13
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
2024-01-10T01:36:49.414Z DRIVER   all queues idle
2024-01-10T01:36:49.952Z SERIAL « 0x012600a80000010039189f037c003d9b2aa3303851f280de17394e8c0177c9f48 (40 bytes)
                                  09000c0007f7f1c
2024-01-10T01:36:49.957Z SERIAL » [ACK]                                                                   (0x06)
2024-01-10T01:36:49.964Z CNTRLR   [Node 057] [~] [Notification] alarmType: 0 => 0                   [Endpoint 0]
2024-01-10T01:36:49.967Z CNTRLR   [Node 057] [~] [Notification] alarmLevel: 0 => 0                  [Endpoint 0]
2024-01-10T01:36:49.970Z DRIVER « [Node 057] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -64 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 124
                                    │ security class:  S2_Authenticated
                                    └─[NotificationCCReport]
                                        notification type:   Water Alarm
                                        notification status: 255
                                        notification state:  Water temperature alarm
2024-01-10T01:36:49.974Z CNTRLR   [Node 057] [~] [Notification] Water Alarm[Water temperature alarm [Endpoint 0]
                                   status]: 0 => 8
```

and this when the water warmed up:

```
2024-01-10T01:38:10.427Z DRIVER   all queues idle
2024-01-10T01:38:10.957Z SERIAL « 0x012400a80000010039169f037e00ca1598e40f5b3570d11fba332d5bc53605d00 (38 bytes)
                                  0c1007f7fdb
2024-01-10T01:38:10.961Z SERIAL » [ACK]                                                                   (0x06)
2024-01-10T01:38:10.965Z CNTRLR   [Node 057] [~] [Notification] alarmType: 0 => 0                   [Endpoint 0]
2024-01-10T01:38:10.968Z CNTRLR   [Node 057] [~] [Notification] alarmLevel: 0 => 0                  [Endpoint 0]
2024-01-10T01:38:10.972Z DRIVER « [Node 057] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -63 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 126
                                    │ security class:  S2_Authenticated
                                    └─[NotificationCCReport]
                                        notification type:   5
                                        notification status: 255
                                        notification state:  idle
                                        event parameters:    0x08
2024-01-10T01:38:10.974Z CNTRLR   [Node 057] [~] [Notification] Water Alarm[Water temperature alarm [Endpoint 0]
                                   status]: 8 => 0
```

I have the whole log if you need more information or if I missed someting

Anto79-ops, 2024-01-10:
by the way, if this matters, this was a water alarm for the water being too cold or below the set low temp setting.

apella12, 2024-01-10:
`HA is dynamically [looking up](https://github.com/home-assistant/core/blob/8181fbab5ccc369a6ca5f0c3f9a45910e7a65c48/homeassistant/components/zwave_js/sensor.py#L733-L735) the valid states`  That is good to know.  I've never used HA, just some of HA JSON "configs" that get published to MQTT and they seem to be fixed (like door open = 22).

I can't decipher the serial, probably because of S2, but it is reading the water temperature alarm state (0x08) as the event.  The format is supposed to be;
[image]
 The serial byte string is hex should contain 71 05 00 00 00 FF 05 08 01 02 (cold).  My guess is the device may only be sending
71 05 00 00 00 FF 05 08 00 00 (just the alarm event, but no additional info), but I can't really tell with the encryption

apella12, 2024-01-10:
`I don't think there's any requirement that event data be provided,`  Agree spec shows state parameters as optional.

I'm afraid I can't help much more because of lack of knowledge about HA and Zwave-js. ;-(
My only parting comment is I'm not sure how Zwave-js is handling the bytes (which are included, which are read as the "state"). To me (0x08, 0x00) taken together are 2048 and maybe could mean "no data" (i.e. no optional state parameter).  Once optional state parameters are included the relevant Byte string increases to 3 (0x08, 0x01, (0x01 || 0x02 || 0x03), so I can't get to 2049, 2050 2050 unless the parameter length byte is skipped. If it just skips to the last Byte then the _2 suffix states make sense.  However, it does seem odd that the spec would add a parameter length byte only to say there is no data, but I haven't researched that.

apella12, 2024-01-10:
FWIW I think your table above is correct and the duplicate entity suffix _2 (3x) are in error (but maybe don't come into play, so do not cause harm?).  Similar to the door there is 0x16, 0x1600 and 0x1601. With my minimal understanding Line 414 moves the alarm from 0x08 to 0x0800 and appends the 01,02,03 to get the values in your table. The "state" that needs to be added is your second line.  Looking at the log posted by @Anto79-ops (the one with cold water) does not show an event parameter, but the return to idle log shows the alarm number as an event.  I have seen this in my Zsniffer logs with a dome leak detector.
[image]

apella12, 2024-01-11:
@kpine First thanks for all the references! They are really helping me learn more about zwave-js and maybe in the future I can actually help solve some issues on github rather than be a pest.

However, one more ;-). By Spec the enum is optional and Zwave-js handles that in line 4456.  So far so good. Since HA is dynamically looking up states provided by the driver why isn't it dynamically picking up "8"?  Both the `valueWithEnum` and `value` are valid.  It seems that is where the problem lies. agree?

AlCalzone, 2024-01-11:
I think @kpine is on point. The door sensor is the only one where the list of states is extended by the event parameters. All others should operate in "replace" mode. So even without the event parameters in the received payload, Z-Wave JS should use one of the defined states (1,2,3) , not the original notification event (8).

apella12, 2024-01-11:
No problem for me; it is up to you guys (obviously ;-) what to do with Z-Wave JS. 

My perspective was just that the state parameter is optional by the ZW spec and deemed "invalid" (only?) by HA.  My concern was just that other device mfgs might opt not to add the state parameter and there are other notifications (e.g. "Meter") with optional state parameters, so the simplest (IHMO) would be add a few lines of code allow the original notification event in HA.  (Not to mention your availability to address in ZWave JS.

As a MQTT user of Zwave JS, either way makes no difference as I can handle any value in the topic once I know what to expect from the device.

PR #6719 (2024-04-04): fix: add workaround for devices that omit notification parameters instead of sending "no data"
fixes: #6584
closes: https://github.com/zwave-js/node-zwave-js/pull/6606
closed 2024-04-04 by zwave-js-bot