zwave-js/zwave-js #6587: Missing device configuration: MH5-4a

rjulius23, 2024-01-03:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

MH5-4A

### Manufacturer ID

0x015f

### Product Type

0xf523

### Product ID

0x7101

### Firmware Version

2.60

### Is the device listed on the Z-Wave Alliance website?

Could not find

### Do you have a manual?

[MH5-4A-23031.pdf](https://github.com/zwave-js/node-zwave-js/files/13823267/MH5-4A-23031.pdf)

rjulius23, 2024-01-03:
So my question is that how can I add the Configuration V4 Parameters to the device json ?

This is what I see after interview:
[image]

And while based on the Manual I can set the params, but would be more user friendly if I can set their name and values accordingly. Can I get please an example how would it look like in the device's json ?

I already copied the MH5-2A.json and renamed it to MH4-4A.json and it works, but these Configuration parameters I have no clue how shall I label them in the json file. 

Thanks in advance.

apella12, 2024-01-09:
Did you delete the paramInformation from the MH5-2A? If not follow the pattern and the manual to adjust the labels and units.

```
	"paramInformation": [
		{
			"#": "1",
			"label": "Temperature Unit",
			"valueSize": 1,
			"defaultValue": 1,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Celsius",
					"value": 0
				},
				{
					"label": "Fahrenheit",
					"value": 1
				}
			]
		},
		{
			"#": "2",
			"label": "Temperature Reporting Threshold",
			"description": "Valid range: 3-255; Variation in temperature to induce automatic report",
			"valueSize": 2,
			"unit": "0.1 °F/C",
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 5,
			"options": [
				{
					"label": "Disable",
					"value": 0
				}
			]
		},
```

PR #7765 (2025-05-12): feat(config): add McoHome thermostats MH4936, MH5-2D and MH5-4A
Fixes #7506
Fixes #7281
Fixes #6587

Device capabilities based on user manuals attached to those issues and those found [here](https://mcohome.com/DownList/1.html). There are several other similar thermostats at that link but I can't find the fingerprints for them.

Templates for parameters added based on those in the database for the MH5-2A. With the templates in place, the config for the MH5-2A will change slightly as the user manual shows Celsius as the default. The MH4936 is the only one of these four thermostats that claims t [... cut]
closed 2025-05-12 by AlCalzone