zwave-js/zwave-js #6588: Pairing issues- Devices going dead while pairing

mohanblaze, 2024-01-04:
Hi Robert,

I am using the chip **ZGM130S** as the controller for the zwave devices to pair and control. While in the pairing process, I am facing issues. 

When I trigger the device after initiating the pairing from zwaveJs-UI, I see that the device addition fails(**node is dead**) within the range of 1 meter. This issue is happening with all devices, i.e., **Power, Battery, and FLIRS**. 

I also see that re-interviewing or refreshing is helping to fetch the device info, but I need to perform it a couple or more times.

I do not know if there's any issue in the zwaveJS UI or with the setting of tx power but I am encountering this issue very frequently. I am attaching the **setting.json** file for reference and the **controller options** screenshot. Let me know, what else is needed to study this issue and to provide us a solution. 

Thank you. 

Regards,
Mohan Krishna

[settings.json](https://github.com/zwave-js/zwave-js-ui/files/13827474/settings.json)
[image]

robertsLando, 2024-01-04:
@AlCalzone any clue what could cause this?

```
2024-01-04T09:57:43.042Z CNTRLR   finished adding node 2:
                                    basic device class:    Routing Slave
                                    generic device class:  Thermostat
                                    specific device class: General Thermostat V2
                                    supported CCs: 
                                    · Basic (0x20)
                                    · Manufacturer Specific (0x72)
                                    · Thermostat Mode (0x40)
                                    · Thermostat Setpoint (0x43)
                                    · Version (0x86)
                                    · Z-Wave Plus Info (0x5e)
                                    · Thermostat Fan Mode (0x44)
                                    · Multilevel Sensor (0x31)
                                    · Association (0x85)
                                    · Configuration (0x70)
                                    · Association Group Information (0x59)
                                    · Transport Service (0x55)
                                    · Device Reset Locally (0x5a)
                                    · Powerlevel (0x73)
                                    · Security (0x98)
                                    · Security 2 (0x9f)
                                    · Supervision (0x6c)
                                    · Firmware Update Meta Data (0x7a)
                                    controlled CCs: 
2024-01-04T09:57:43.046Z CNTRLR » [Node 002] Assigning SUC return route...
2024-01-04T09:57:43.047Z CNTRLR » [Node 002] Deleting SUC return route...
2024-01-04T09:57:43.060Z SERIAL » 0x01060055000205ab                                                   (8 bytes)
2024-01-04T09:57:43.062Z DRIVER » [Node 002] [REQ] [DeleteSUCReturnRoute]
                                    payload: 0x000205
2024-01-04T09:57:43.083Z SERIAL « [ACK]                                                                   (0x06)
2024-01-04T09:57:43.086Z SERIAL « 0x0104015501ae                                                       (6 bytes)
2024-01-04T09:57:43.088Z SERIAL » [ACK]                                                                   (0x06)
2024-01-04T09:57:43.090Z DRIVER « [RES] [DeleteSUCReturnRoute]
                                    was executed: true
2024-01-04T09:57:43.124Z SERIAL « 0x01210055050139000000007f7f7f7f7f00010400000000020300007f7f7f7f7f0 (35 bytes)
                                  000b2
2024-01-04T09:57:43.126Z SERIAL » [ACK]                                                                   (0x06)
2024-01-04T09:57:43.128Z DRIVER « [REQ] [DeleteSUCReturnRoute]
                                    callback id:     5
                                    transmit status: NoAck
2024-01-04T09:57:43.142Z SERIAL » 0x01060051000206ac                                                   (8 bytes)
2024-01-04T09:57:43.144Z DRIVER » [Node 002] [REQ] [AssignSUCReturnRoute]
                                    payload: 0x000206
2024-01-04T09:57:43.147Z SERIAL « [ACK]                                                                   (0x06)
2024-01-04T09:57:43.150Z SERIAL « 0x0104015101aa                                                       (6 bytes)
2024-01-04T09:57:43.152Z SERIAL » [ACK]                                                                   (0x06)
2024-01-04T09:57:43.154Z DRIVER « [RES] [AssignSUCReturnRoute]
                                    was executed: true
2024-01-04T09:57:43.231Z SERIAL « 0x01210051060151000000007f7f7f7f7f00010600000000020300007f7f7f7f7f0 (35 bytes)
                                  000df
2024-01-04T09:57:43.233Z SERIAL » [ACK]                                                                   (0x06)
2024-01-04T09:57:43.236Z DRIVER « [REQ] [AssignSUCReturnRoute]
                                    callback id:     6
                                    transmit status: NoAck
2024-01-04T09:57:43.455Z SERIAL » 0x011000a900010002029f04250000000007fe                              (18 bytes)
2024-01-04T09:57:43.459Z DRIVER » [Node 002] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      7
                                  └─[Security2CCKEXGet]
2024-01-04T09:57:43.474Z SERIAL « [ACK]                                                                   (0x06)
2024-01-04T09:57:43.476Z SERIAL « 0x010401a90152                                                       (6 bytes)
2024-01-04T09:57:43.478Z SERIAL » [ACK]                                                                   (0x06)
2024-01-04T09:57:43.481Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2024-01-04T09:57:47.645Z SERIAL « 0x011d00a90701019c007f7f7f7f7f00010700000000020400000000000000af    (31 bytes)
2024-01-04T09:57:47.658Z SERIAL » [ACK]                                                                   (0x06)
[... 141 lines cut ...]
                                  │ transmit options: 0x25
                                  │ callback id:      10
                                  └─[NoOperationCC]
2024-01-04T09:58:04.589Z SERIAL « [ACK]                                                                   (0x06)
2024-01-04T09:58:04.593Z SERIAL « 0x010401a90152                                                       (6 bytes)
[... cut ...]
                                    measured noise floor:   0 dBm
2024-01-04T09:58:08.769Z CNTRLR   [Node 002] The node did not respond after 1 attempts, it is presumed dead
2024-01-04T09:58:08.772Z CNTRLR   [Node 002] ping failed: The node did not acknowledge the command (ZW0204)
2024-01-04T09:58:08.774Z CNTRLR   [Node 002] failed to refresh values for Thermostat Mode, endpoint 0: The messa
                                  ge cannot be sent because node 2 is dead (ZW0202)
2024-01-04T09:58:08.779Z DRIVER   all queues idle
```

AlCalzone, 2024-01-04:
Did you change the power values, especially the measured at 0? If so, did someone tell you which values to use?

mohanblaze, 2024-01-05:
Hi @AlCalzone ,

Thanks for your response. 

Yes, we updated the values of the measured at 0 to 3.3. PFA screenshot for reference.

And we are using **ZGM130S** , which is a silabs controller(700 series). Let me know if anything is needed.
 

Regards,
Mohan Krishna

[image]
[image]

mohanblaze, 2024-01-09:
Hi @AlCalzone,

Do you happen to have any updates on the above data? Can you please help us to sort out this?

Regards,
Mohan Krishna

AlCalzone, 2024-01-09:
https://github.com/zwave-js/node-zwave-js/issues/6521

apella12, 2024-01-09:
One thing that might help is to upgrade the firmware on the Zstick.  I see from the picture it is 7.15.4.  Might want to read this https://zwave-js.github.io/node-zwave-js/#/troubleshooting/otw-upgrade

mohanblaze, 2024-01-10:
Hi @apella12,

Thanks for your response.

Yes, the thermostat is connected to a 5v DC input. The issue is not with a single device, and all powered devices are working in the same manner. When the pairing is initiated, we trigger the node info button to release **NodeInfo** from the device. Thereafter, we noticed the device is paired as dead and once we refresh CC or re-interview the nodeId then it'll pull the info from the device. It just stopped while pairing, as you said like a battery device.

Btw, Is there any problem with the power settings? I used 9.9, 3.3 dbm as the values for the normal power level & measured power at 0 dbm. Help us to resolve this issue.

Regards,
Mohan Krishna

apella12, 2024-01-10:
As this is happening with all devices it could be the controller.  I would update the firmware to 7.18.3 That what I am using.  As to the power I don't know.  I have gotten conflicting advice from mfg, but my 700 is (6, 0) in USA).  Your values could be too high.

AlCalzone, 2024-01-12:
Unfortunately, the logs can't tell us why there is a connectivity problem with the device, but there is one. A few ideas:
- The inclusion **might** (not sure) be happening at 40 kbps and then for normal communication the controller attempts to use 100 kbps, which is channel 0. That channel has a much higher background noise than the others (-87 dBm vs. -108 dBm), so there might be too much noise for reliable communication. Is there another device nearby that could cause this RF noise?
- I'm still not sure which controller you have (the actual device, not the chip). Most 700 series controllers are indistinguishable from software. Is this a GPIO module or a USB stick? If it's a USB stick, did you put it on an [extension cord](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting)?
- Where did you get the values for the power setting from? Wrong values may cause connectivity issues.
- Are both the controller and the devices using the same RF region (Europe)?

closed 2024-07-09 by AlCalzone