zwave-js/zwave-js #6597: Missing device configuration: Utilitech Wireless Indoor Siren #TSE07-1

cirrusflyer, 2024-01-09:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Utilitech

### Manufacturer ID

0

### Product Type

0

### Product ID

0

### Firmware Version

0

### Do you have a manual?

[tse071.pdf](https://github.com/zwave-js/node-zwave-js/files/13868946/tse071.pdf)

apella12, 2024-01-09:
It will be impossible to add this device without TYPE, ID. Are you sure about this assertion?
`The device was interviewed completely and has no missing or undefined IDs`

cirrusflyer, 2024-01-09:
I was able to add it to Scrypted and it's showing me this, so maybe it's just another manufacturer:

https://devices.zwave-js.io/?jumpTo=0x0060:0x000c:0x0001:1.0

It displays "type" as a "light".

apella12, 2024-01-10:
Interesting.  As you can see that resolves to an Everspring Siren.  It has no parameters and just one Association group. It is not a "light", but from poking around should show a Switch_binary (like a light) and a notification/alarm channel.  It is in the config DB, so should be showing up? (se812.json)
```
{
	"manufacturer": "Everspring",
	"manufacturerId": "0x0060",
	"label": "SE812",
	"description": "Siren",
	"devices": [
		{
			"productType": "0x000c",
			"productId": "0x0001"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"associations": {
		"1": {
			"label": "Control/Report Group",
			"maxNodes": 5,
			"isLifeline": true
		}
	}
}
```
What versions are you on? <html>
<body>

Inclusion Information | Pressing link key 3 times within 1.5 seconds will enter inclusion mode
-- | --

</body>
</html>

cirrusflyer, 2024-01-10:
Sorry, I should have been more clear.  I got it working in Scrypted as a light, and when activated it activates the siren and strobe.

apella12, 2024-01-10:
I do not know what Scrypted is?  Generally, how it works is during inclusion the device advertises the mfgID, TypeID and ProductID (in this case 0x0060, 0x000c & 0x0001) and Zwave-js uses that info to search the DB of JSON devices. The actual brand name and Device labeling are irrelevant.  Many devices (particularly Chinese) are clones or direct copies.  With that as context, have you tried to include the device? It will be labeled Everspring SE812, but should work fine.

cirrusflyer, 2024-01-10:
Yes, it's working find as a light device that allows me to trigger the strobe and siren.  Thanks

apella12, 2024-01-10:
FYI- It is your home automation software that is labeling it a light, not zwave-js. In ZWave terms it is a binary switch (used by a light, but also other things. Close issue if you are ok.

closed 2024-10-30 by AlCalzone