zwave-js/zwave-js #6607: Consistent timeouts trying to upgrade firmware on two identical devices

mattster98, 2024-01-15:
I've tried dozens of times to upgrade the firmware on these two Aeotec ZW111 nano dimmers.  They've been a little weird lately already, which is why I was interested in a firmware upgrade.  By weird, I mean they've gone "offline" in some way, not officially, but unable to control and require a hard power off (at the breaker box) in order to get them operable again.  They're working fine now and I've tried hard power offs to resolve this issue.  The controller is < 1m away from these devices.

Debug log of the last attempt (Node 28) attached.  Snip of the last couple successful fragments up to the timeout pasted below.

It's a pretty consistent pattern, other than the amount of progress it makes varies from <1% to >15% before it says "timed out".

I've tried on 3.0.2 and 3.1.0.  Hardware is Razberry 7 pro.

[zwavejs_2024-01-13 (3).log](https://github.com/zwave-js/zwave-js-ui/files/13929587/zwavejs_2024-01-13.3.log)

2024-01-13T21:08:10.528Z CNTRLR » [Node 028] Sending firmware fragment 28 / 6240
2024-01-13T21:08:10.532Z DRIVER   all queues busy
2024-01-13T21:08:10.535Z SERIAL » 0x011000a90001001c02984025000000009337                              (18 bytes)
2024-01-13T21:08:10.535Z DRIVER » [Node 028] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      147
                                  └─[SecurityCCNonceGet]
2024-01-13T21:08:10.544Z SERIAL « [ACK]                                                                   (0x06)
2024-01-13T21:08:10.545Z SERIAL « 0x010401a90152                                                       (6 bytes)
2024-01-13T21:08:10.545Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:08:10.546Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2024-01-13T21:08:10.555Z SERIAL « 0x011d00a99300000100d37f7f7f7f00000300000000030100007f7f7f7f7f74    (31 bytes)
2024-01-13T21:08:10.556Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:08:10.557Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            147
                                    transmit status:        OK, took 10 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    routing scheme:         LWR
                                    ACK RSSI:               -45 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2024-01-13T21:08:10.564Z SERIAL « 0x011800a8000001001c0a9880ad61141f709c1f5000d3007f7ff7              (26 bytes)
2024-01-13T21:08:10.565Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:08:10.566Z DRIVER « [Node 028] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -45 dBm
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xad61141f709c1f50
2024-01-13T21:08:10.569Z SERIAL » 0x013c00a90001001c2e988121f8559ad2ba2fa2f026fe628638f9ca335ef6c5057 (62 bytes)
                                  1f6518e523f0f7c8b903f9a6047add51748bf06a8f3c025000000009444
2024-01-13T21:08:10.570Z DRIVER » [Node 028] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      148
                                  └─[SecurityCCCommandEncapsulation]
                                    │ nonce id: 173
                                    └─[FirmwareUpdateMetaDataCCReport]
                                        report #: 28
                                        is last:  false
2024-01-13T21:08:10.580Z SERIAL « [ACK]                                                                   (0x06)
2024-01-13T21:08:10.581Z SERIAL « 0x010401a90152                                                       (6 bytes)
2024-01-13T21:08:10.581Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:08:10.583Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2024-01-13T21:08:10.612Z SERIAL « 0x011d00a99400000300d37f7f7f7f00000300000000030100007f7f7f7f7f71    (31 bytes)
2024-01-13T21:08:10.613Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:08:10.615Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            148
                                    transmit status:        OK, took 30 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    routing scheme:         LWR
                                    ACK RSSI:               -45 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2024-01-13T21:08:10.617Z DRIVER   all queues idle
2024-01-13T21:08:10.619Z SERIAL « 0x011800a8000001001c0a9880ad61141f709c1f5000d3007f7ff7              (26 bytes)
2024-01-13T21:08:10.621Z SERIAL » [ACK]                                                                   (0x06)
[... 148 lines cut ...]
2024-01-13T21:09:01.944Z DRIVER   all queues idle
2024-01-13T21:09:31.926Z DRIVER   all queues busy
2024-01-13T21:09:31.931Z SERIAL » 0x0103003bc7                                                         (5 bytes)
2024-01-13T21:09:31.933Z DRIVER » [REQ] [GetBackgroundRSSI]
2024-01-13T21:09:31.936Z SERIAL « [ACK]                                                                   (0x06)
2024-01-13T21:09:31.938Z SERIAL « 0x0107013b9c98987f21                                                 (9 bytes)
2024-01-13T21:09:31.940Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:09:31.943Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -100 dBm
                                    channel 1: -104 dBm
                                    channel 2: -104 dBm
2024-01-13T21:09:31.947Z DRIVER   all queues idle
2024-01-13T21:10:01.927Z DRIVER   all queues busy
2024-01-13T21:10:01.936Z SERIAL » 0x0103003bc7                                                         (5 bytes)
2024-01-13T21:10:01.937Z DRIVER » [REQ] [GetBackgroundRSSI]
2024-01-13T21:10:01.940Z SERIAL « [ACK]                                                                   (0x06)
2024-01-13T21:10:01.941Z SERIAL « 0x0107013b9c99997f21                                                 (9 bytes)
2024-01-13T21:10:01.944Z SERIAL » [ACK]                                                                   (0x06)
2024-01-13T21:10:01.948Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -100 dBm
                                    channel 1: -103 dBm
                                    channel 2: -103 dBm
2024-01-13T21:10:01.953Z DRIVER   all queues idle
2024-01-13T21:10:12.647Z CNTRLR   [Node 028] Firmware update timed out
2024-01-13T21:10:12.649Z CNTRLR « [Node 028] Firmware update (part 1 / 1) failed with status Error_Timeout

mattster98, 2024-03-08:
I've replaced these devices.

robertsLando, 2024-03-08:
@mattster98 I'm sorry for the missing response but @AlCalzone is still on parental leave and will be back on April 🙏🏼

mattster98, 2024-03-08:
No worries - I just didn't want someone poring over my logs and then ask me to try something.