zwave-js/zwave-js #6612: High CPU usage with docker

Giel538, 2024-01-18:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.6.2.6e369a1

### ZwaveJS version

12.4.1

### Describe the bug

I am running all my smart home stuff inside a debian virtual machine. It was almost using 2% cpu power steady for months. Since the update from Zwavejs UI to 9.6.0 my cpu usage is increased to 3.1%. I know it is still low in absolute value but it is still almost 35%! The vm is using 2 cores from my intel i5 13500 cpu.  May be other people who are using a rpi or similar systems are have more problems with the increased cpu usage.

ZwaveJS is using the most cpu power from all my docker containers. Even more then Home Assistant or Zigbee2mqtt (which has a lot more devices). 

I checked the change log but i don't know what is changed what can cause a higher cpu usage

### To Reproduce

I restarted the container a few times but it is still the same. When i stop the container the cpu usage is back to 2%. 

### Expected behavior

I expect after a "small" update that the cpu usage is not changing this much. 

### Additional context

I am using AEON Labs | Z‐Stick Gen5 USB Controller but it is connected via ser2net via another computer. 

This is my docker stack:

```
services:
  zwavejs2mqtt:
    container_name: zwavejs2mqtt
    image: zwavejs/zwavejs2mqtt:latest
    restart: unless-stopped
    tty: true
    stop_signal: SIGINT
    environment:
#      - SESSION_SECRET=mysupersecretkey
      - TZ=Europe/Amsterdam
#      - ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db      
#    devices:
#      - /dev/serial/by-id/usb-0658_0200-if00:/dev/zwave
    volumes:
      - /home/xxxxx/data/zwavejs2mqtt/store:/usr/src/app/store
    ports:
      - '8091:8091' # port for web interface
      - '3000:3000' # port for Z-Wave JS websocket server
```

Giel538, 2024-01-18:
> > just throwing some ideas, could it be the jsonl db?
> 
> While this does loop quite a bit, most of the time should be spent sleeping. And it's only 50 of ~18000 total ticks.
> 
> @t-o-o-m you could try editing /usr/src/app/node_modules/@alcalzone/jsonl-db/build/lib/db.js line 592 for a test:
> 
> ```diff
> -         const sleepDuration = 20; // ms
> +         const sleepDuration = 250; // ms
> ```
> 
> for example.

Hi.

I can also confirm that this is working for me as well.

closed 2024-01-25 by zwave-js-bot

robertsLando, 2024-01-25:
Hi guys, could you give a try to `master` tag?

t-o-o-m, 2024-01-25:
Can confirm low idle CPU usage on the `master` branch. Great work @AlCalzone @robertsLando!