zwave-js/zwave-js #6622: Fibaro FGPWE-102 ZW5 device (wall plug) not recognized  as a FGWP102

brannou, 2024-01-22:
Hello,
How can I add this device to zwave-js-ui? It doesn't work with the existent FGWP102.json file.

FGWP102 exists in the DB browser:
Fibargroup / Metered Wall Plug Switch
Firmware Range: All Versions
Product Manufacturer: 0x010f
Product Identifiers: 0x0602:0x1001, 0x0602:0x1003, 0x0602:0x1004, 0x0602:0x4001, 0x0602:0x4003

But I guess my FGWPE-102 ZW5 has a different identifier (which was supposed to be 0x0602:0x1001)
Mine is recognized as "Unknown manufacturer 0xXXXX" without any socket management. However its zwave gateway works fine! 

FYI, I got it working in Domoticz/OpenZwave by adding a FGWPEZW5.xml file but I don't know how to do the same in zwave-js-ui.

Does anyone have a solution here??

Thanks,
brannou

AlCalzone, 2024-01-22:
> Mine is recognized as "Unknown manufacturer 0xXXXX"

This means there were communication issues during the interview and it's not fully recognized.
Try re-interviewing.

brannou, 2024-01-22:
Thanks ALCalzone for your answer but I already tried this several times. Is there a way to query this Fibaro plug manually and check its identifiers fit the json file?
[image]

AlCalzone, 2024-01-22:
I suspect it was included using encryption (Security S0), but you didn't specify the key from OZW in the Z-Wave configuration.

brannou, 2024-01-22:
These ones??
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/14014762/zwave-js-ui-store.zip)

AlCalzone, 2024-01-22:
Okay, that's an odd issue. The device does not report support for Version CC (needed to query the specific versions of each CC the device supports), nor does it report support for the Manufacturer Specific CC (needed to identify the device). The latter is the bigger problem, as it makes it impossible to identify the device - at least with Z-Wave JS's current implementation.

I suspect the device does support these CCs and just reports the wrong information. Please try excluding it, factory resetting it, then including it again. Maybe that solves your issue already. If not, I'll have to implement a workaround when I'm back from [parental leave](https://github.com/zwave-js/node-zwave-js/issues/6521).

This was probably no issue in OZW, as it hardcoded a lot of information about devices, unlike Z-Wave JS which relies mostly on discovering devices' capabilities.

brannou, 2024-01-22:
Hi ALCalzone, After a real reset procedure (well followed by myself...) it works amazingly well! Fantastic
Many thanks for your help and enjoy your parental leave!
brannou