zwave-js/zwave-js #6635: Missing device configuration: Ring Glass Break Sensor EU 0x0401

candrea77, 2024-01-23:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Ring Glass Break Sensor EU

### Manufacturer ID

0x0346

### Product Type

0x0a01

### Product ID

0x0401

### Firmware Version

1.11.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4560

### Do you have a manual?

Hi @AlCalzone , can you please approve [PR 6590](https://github.com/zwave-js/node-zwave-js/pull/6590) for missing device fingerprint ?

apella12, 2024-01-23:
Have you tried to put the modified json file in your store/config folder and reinterview your device.  That should get you going.

candrea77, 2024-01-23:
Yep. 
It works fine. 
I'm just try to understand when the PR will be merged and if there is something that I've to do (it's my first PR ... so If I have missed something .... please let me know)

[image]

apella12, 2024-01-23:
Did you see the tagged item about parental leave? I think the worst-case scenario is April.  I have several open PRs myself.  IMO yours should sail through.

candrea77, 2024-01-24:
> Did you see the tagged item about parental leave? 
Awh ... didn't see it !!!
This explain all of this

> I think the worst-case scenario is April. I have several open PRs myself. IMO yours should sail through.
Yep .... I was worried about have done something wrong!
So no prbl .... it's a very very low priority PR

PR #6590 (2024-04-02): fix(config): add fingerprint for Ring Glass Break Sensor EU
Additional fingerprint of Ring Glass Break Sensor device for the UE version (0x0346 0x0a01-0x0401)

fixes: #6635
closed 2024-04-02 by AlCalzone