zwave-js/zwave-js #6645: Update project description to include 800-series devices

TheKorn2, 2024-01-29:
On project main landing page https://github.com/zwave-js , description currently reads:

"Built on Node.js and compatible with all known 500 and 700 series Z-Wave controller hardware adapters, the Z-Wave JS driver runs on almost anything with a little bit of computing power and a serial port."

Should be updated to read:

Built on Node.js and compatible with all known 500, 700 and 800 series Z-Wave controller hardware adapters, the Z-Wave JS driver runs on almost anything with a little bit of computing power and a serial port.

I know it's trivial, but might as well squash the easy ones!

AlCalzone, 2024-01-29:
Go for it!

TheKorn2, 2024-04-09:
@AlCalzone  It was? Still says 700 here.

AlCalzone, 2024-04-09:
Ohhh, I only updated the Readme of this repo. Not the organization Readme.

closed 2024-04-12 by AlCalzone

TheKorn2, 2024-04-12:
Looks good this time! :D