zwave-js/zwave-js #6646: keep having to reset priority route

phemmer, 2024-01-29:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

9.8.1

### ZwaveJS version

12.4.2

### Describe the bug

I have a zwave switch inside a metal enclosure that severely limits the signal. However I have another zwave device 8 inches away outside the metal enclosure that can act as a relay. For some reason zwave isn't able to automatically figure out that it should use this device as a relay, and I have to manually set a priority route. The problem is that it seems this doesn't persist. It only works for a short while, and then goes dead until I set the route again.

### To Reproduce

Set priority route for device B through device A.  
Wait a little bit, turn device B on/off a few times, whatever.  
Watch as device A goes dead until the priority route is set again.

### Expected behavior

Device should continue working without having to reset priority route.

### Additional context

After setting the priority route, it shows in the network graph. However in the "Node properties" panel, it says "None" under "Priority route".

In the provided logs, the device that keeps failing is "Pool light switch" (Node 12), and I'm setting the priority route through "Pool pump relay" (Node 10). I also have to set the speed at 9.6Kbps as anything faster is unreliable.

The following times were the notable events (approximated as I was looking at the clock to keep notes when event occurred) :
* 22:02:30 - Died when turning off
* 22:02:45 - Issued ping that failed
* 22:03:15 - Added priority route
* 22:03:20 - Turned off successfully
* 22:03:55 - Dead again when turning on
* 22:04:30 - Added priority route
* 22:04:40 - Turned on successfully

[z-ui_2024-01-28.log](https://github.com/zwave-js/zwave-js-ui/files/14079201/z-ui_2024-01-28.log)
[zwavejs_2024-01-28.log](https://github.com/zwave-js/zwave-js-ui/files/14079202/zwavejs_2024-01-28.log)

AlCalzone, 2024-02-04:
I'm not sure which controller you have, but it looks like that doesn't support setting priority routes. After setting the route (seemingly successfully), it immediately responds that the route to node 12 is the last working route:
```
2024-01-29T03:04:30.075Z DRIVER « [RES] [GetPriorityRoute]
                                    node ID:     12
                                    route kind:  LWR
                                    repeaters:   10
                                    route speed: 9k6
```
`route kind` should be `Application` for the priority route.

That said, I think your problem is actually the excessive reporting of node 10. Every 4-5 seconds, it sends 2 meter reports for V and A. Sometimes unchanged, sometimes changes of 1/100 V/A.
Z-Wave doesn't deal with this kind of traffic well. Effectively Node 10 takes away most of the transmit time from other devices. When devices are having trouble reaching their destination through or past node 10, they will repeat commands, which can cause a lot of traffic we're not even seeing in the logs.

You should take a look at https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=optimizing-the-reporting-configuration and apply it. Also, if you haven't already: [put your controller on a USB extension](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting), assuming it's a USB stick.
I wouldn't be surprised if changing these two things solves your issue(s).

Interestingly, this even causes problems communicating with node 10 itself:
```
2024-01-29T03:00:01.268Z DRIVER » [Node 010] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      19
                                  └─[BinarySwitchCCGet]
2024-01-29T03:00:01.280Z DRIVER « [RES] [SendData]
                                    was sent: true

...

2024-01-29T03:00:05.874Z DRIVER « [Node 010] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      V
                                      rate type:  Consumed
                                      value:      247.653
                                      time delta: 0 seconds

...

2024-01-29T03:00:09.993Z DRIVER « [REQ] [SendData]
                                    callback id:     19
                                    transmit status: NoAck
2024-01-29T03:00:09.998Z CNTRLR   [Node 010] The node did not respond after 1 attempts, it is presumed dead
2024-01-29T03:00:10.005Z CNTRLR   [Node 010] The node is now dead.
```

phemmer, 2024-02-09:
Z-Wave stick was [this](GoControl CECOMINOD016164 HUSBZB-1 USB Hub) which has a ton of different names (Physically says "Nortek". The amazon listing says "GoControl". The device when plugged in identifies as "Silicon Labs").  
And yes, it was used on a USB extension cord.

In any case, I replaced the stick with Zooz ZST39. I don't know if priority routes work correctly with it as it appears no longer necessary. Even inside with the device in the metal enclosure the stick can communicate directly with the device at full 100kbit.

So I'm going to go ahead and close this.