zwave-js/zwave-js #6650: CT100 ZWave thermostats report extra values after reboot

ken-wolfe, 2024-02-01:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

9.8.1

### ZwaveJS version

12.4.2

### Describe the bug

Occasionally my thermostats (I have 2 of them, both CT100), report extra values (2 additional full thermostats, 1 additional air temp sensor). This seems to happen after rebooting my Home Assistant computer, restarting HA by itself does not seem to trigger the issue. Re-interviewing the node for each thermostat, causes these additional values to get removed.

The log files below are a copy and paste of the Node-JS UI "Debug" screen for node 30, 1 from before I reinterviewed the node and one from after. The after does not contain the extra values.

[node30-after-interview.txt](https://github.com/zwave-js/zwave-js-ui/files/14073357/node30-after-interview.txt)
[node30-before-interview.txt](https://github.com/zwave-js/zwave-js-ui/files/14073358/node30-before-interview.txt)

Here is a full Driver log from Node-JS UI for node 5 and node 30 (my thermostats) which should show before I re-interviewed, the re-interview, a restart of HA, a reboot of the HA machine (which then causes the problem to repro), and a re-interview of the devices (which removes the problem until the next reboot of the machine).

[zwavejs_2024-01-27(1).log](https://github.com/zwave-js/zwave-js-ui/files/14073370/zwavejs_2024-01-27.1.log)

Let me know if there is anything more that you need from me. Now that I know how to reproduce the problem, I should be able to get additional logs very easily.

### To Reproduce

Have Home Assistant
Have ZWave-JS UI installed
Re-interview one of the thermostat nodes, notice that extra values go away
Reboot Home Assistant computer
Notice that extra values have returned.

### Expected behavior

No extra values for the thermostat nodes appear. There should be only 1 thermostat control per node.

### Additional context

Sometimes the extra air temp sensor does not show up. Not sure why that is.

ken-wolfe, 2024-02-02:
@AlCalzone Is there something I should try? Sorry it’s not clear to me what to do next.

AlCalzone, 2024-02-02:
No, you'll have to wait until I'm back from [parental leave](https://github.com/zwave-js/node-zwave-js/issues/6521). I'll try to reproduce this in a test case, so I can fix it for good.

ken-wolfe, 2024-02-02:
Sorry to have bothered you on parental leave. Enjoy the time with your child (or children). If there are any additional logs that would be helpful, I’m happy to gather them. Just let me know.

AlCalzone, 2024-02-02:
It's fine. I check in here when I do have time to kill.

AlCalzone, 2024-04-15:
This should be fixed since v12.5.4 - might need a re-interview though.
If not, please let me know.

txwindsurfer, 2024-04-16:
I am on Core Version 2024.4.3, OS 12.2 and continue to have this problem whenever a new version of Z-Wave-JS-UI is installed (perhaps when restarted).  After re-interview, the extra entities are unavailable.  If I remove the unavailable entities, they reappear with the next Z-Wave-JS-UI update/restart, and I must re-interview all the thermostats again (I have 5 CT101 and CT100).  Perhaps I should not remove the unavailable extra entities.

Previously reported in #6184 and the fix in #6185 fixed the problem with re-interview but not still happens with an update/restart.

Nothing urgent, just a bother.

AlCalzone, 2024-04-16:
Which version of Z-Wave JS UI are you using?

txwindsurfer, 2024-04-16:
zwave-js-ui: 9.10.2
zwave-js: 12.5.2

AlCalzone, 2024-04-17:
Like I said, the fix (https://github.com/zwave-js/node-zwave-js/pull/6760) is in version 12.5.**4**, so you don't have it yet.

txwindsurfer, 2024-04-22:
Just upgdated:
zwave-js-ui: 9.11.1
zwave-js: 12.5.5
And the issue is still present.  Extra entities that must be removed with a Re-interview.
[image]
After Re-interview:
[image]

AlCalzone, 2024-04-22:
Ok then I need logs please.

ken-wolfe, 2024-04-22:
Here are the logs.

I started with system running, and the extra items are deleted. I turned on "Debug" level logging and logged it to a file. The logs should contain the following actions: I restarted Home Assistant. The zombie items came back, I re-interviewed each device and deleted the zombie items after the re-interview. Let me know if you'd like a new bug or any additional logs or information.

[zwavejs_2024-04-22.log](https://github.com/zwave-js/node-zwave-js/files/15066677/zwavejs_2024-04-22.log)

ken-wolfe, 2024-04-22:
In case it matters, I delete the zombie items from the Home Assistant UI. If I should be removing them from elsewhere, let me know.

AlCalzone, 2024-04-23:
You cannot really delete them, as HA just creates entities for what Z-Wave JS exposes. I need a bit more info I'm afraid.

1. enable **application logs** in the settings if you haven't already
2. go to the log tab, pop out the logs into a new window
3. go to the control panel tab, click the menu in the bottom right corner -> advanced -> driver function.
4. Paste the following code there, and replace the 2 in line 2 with the node ID of your node. **Do not press RUN yet.**
	```ts
	const { logger } = this;
	const node = driver.controller.nodes.get(   2   ); // <-- REPLACE 2 WITH YOUR NODE ID
	
	for (const ep of node.getAllEndpoints()) {
		logger.info(`Endpoint ${ep.index}, CCs: ${JSON.stringify([...ep.getCCs()], null, 2)}`);
	}
	
	const valueIDs = node.getDefinedValueIDs();
	logger.info(`value IDs: ${JSON.stringify(valueIDs, null, 2)}`);
	```
5. In the log window, press CLEAR
6. In the Driver function window, press RUN
7. In the log window, press STOP, copy the entire log there and share it here.

ken-wolfe, 2024-04-25:
I'm going to follow the directions you provided, but also want to make sure I'm doing the right thing.

> enable application logs in the settings if you haven't already

I'm assuming this means turn on "Debug" logging in Z-Wave JS. So this is what I have done.

> go to the log tab, pop out the logs into a new window

Again, I'm assuming that this is in Z-Wave JS

> go to the control panel tab, click the menu in the bottom right corner -> advanced -> driver function.

Same assumption as above, in Z-Wave JS

I did it twice, one for each node in case there is something to be learned there. Please let me know if you need any additional things. I'm happy to help.

Here is the output for node 5:
2024-04-24 22:23:27.888 DEBUG SOCKET: Event ZWAVE_API emitted to gnmplGX5UvE6UmxxAAAR
2024-04-24 22:23:27.893 INFO Z-WAVE: Calling api driverFunction with args: [
  'const { logger } = this;\n' +
    'const node = driver.controller.nodes.get(   5   ); // <-- REPLACE 2 WITH YOUR NODE ID\n' +
    '\n' +
    'for (const ep of node.getAllEndpoints()) {\n' +
    '\tlogger.info(`Endpoint ${ep.index}, CCs: ${JSON.stringify([...ep.getCCs()], null, 2)}`);\n' +
    '}\n' +
    '\n' +
    'const valueIDs = node.getDefinedValueIDs();\n' +
    'logger.info(`value IDs: ${JSON.stringify(valueIDs, null, 2)}`);',
  [length]: 1
]
2024-04-24 22:23:27.896 INFO Z-WAVE: Endpoint 0, CCs: [
  [
    114,
    {
      "isSupported": true,
      "isControlled": false,
      "secure": false,
      "version": 1
    }
  ],
  [
    64,
    {
      "isSupported": true,
      "isControlled": false,
      "secure": false,
      "version": 2
    }
  ],
  [
    67,
    {
      "isSupported": true,
      "isControlled": false,
      "secure": false,
      "version": 2
    }
  ],
  [
    134,
    {
      "isSupported": true,
      "isControlled": false,
      "secure": false,
      "version": 1
    }
  ],
  [
    129,
    {
[... 894 lines cut ...]
  },
  {
    "commandClassName": "Multilevel Sensor",
    "commandClass": 49,
    "endpoint": 2,
    "property": "Humidity",
    "propertyName": "Humidity"
  },
  {
    "commandClassName": "Thermostat Mode",
    "commandClass": 64,
    "endpoint": 2,
    "property": "mode",
    "propertyName": "mode"
  },
  {
    "commandClassName": "Thermostat Mode",
    "commandClass": 64,
    "endpoint": 2,
    "property": "manufacturerData",
    "propertyName": "manufacturerData"
  }
]
2024-04-24 22:26:43.383 INFO Z-WAVE: Success zwave api call driverFunction undefined
2024-04-24 22:26:47.667 INFO APP: GET /health/zwave 301 0.742 ms - 191

ken-wolfe, 2024-04-25:
BTW, running the script in the driver reproduced the issue.

ken-wolfe, 2024-05-06:
@AlCalzone do you need any additional information from me?

ken-wolfe, 2024-06-05:
Just wondering if there are any updates here or if I can help in any way. Thanks!

AlCalzone, 2024-06-05:
Sorry, I have to admit I forgot about this. I still don't fully understand what's happening - I might have to simulate the device and try to figure it out that way.

Could you repeat what you did in https://github.com/zwave-js/node-zwave-js/issues/6650#issuecomment-2076402400, but immediately after re-interviewing a device? One of the two is enough.

ken-wolfe, 2024-06-09:
So I've got some good news. I was about to repro the issue (which had repro'd just a few days ago) when I noticed a new update to zwave-js for Home Assistant. I took the update (which would normally cause the issue), but it did not. So, I rebooted the entire system, which would also normally cause the issue to repro, and it did not. So, perhaps the last update for zwave-js did not contain your fix and the one I took today did. I think this can be closed now.

AlCalzone, 2024-06-09:
Hmm according to your logs you should have had the fixed version already.
But feel free to reopen this when it happens again.

closed 2024-06-09 by AlCalzone