zwave-js/zwave-js #6658: Missing device configuration: Fibaro FGMS-001

Jac-Kok, 2024-02-03:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Fibaro FGMS-001

### Manufacturer ID

0xXXXX

### Product Type

0xXXXX

### Product ID

0xXXXX

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3336?selectedFrequencyId=-1

### Do you have a manual?

https://manuals.fibaro.com/motion-sensor/
https://manuals.fibaro.com/content/manuals/en/FGMS-001/FGMS-001-EN-T-v2.1.pdf

AlCalzone, 2024-02-03:
We have this: https://devices.zwave-js.io/?jumpTo=0x010f:0x0800:0x1001:0.0

Since you didn't provide any IDs, I'm assuming your device wasn't interviewed completely.

closed 2024-02-03 by AlCalzone

Jac-Kok, 2024-02-05:
Thank you for your response, You are absolutely right, I was not aware that this could take up to 24 hours for my wireless zwave modules. It now works properly.

AlCalzone, 2024-02-05:
You can always speed up the process by manually waking the devices