zwave-js/zwave-js #6674: Please expose endpoint 0 for Vision ZL7432

TheKorn2, 2024-02-13:
The Vision ZL7432 has three end points, 0, 1, and 2.  0 and 1 control the same physical device (device 1), but only endpoint 0 supports controller notifications on local changes, while end point 1 does not. See https://opensmarthouse.org/zwavedatabase/112 

(Don't ask me why for the love of everything holy there are two endpoints on the same device that do the same thing with varying capabilities.)

Right now only endpoints 1 and 2 are exposed, meaning any local changes (i.e. wall switch changes) are unreported in the UI, forcing polling. Polling bad, mmmmmkay?

I'd like for endpoint 0 to be exposed along with 1 and 2. Really you could hide endpoint 1 entirely and I'd be fine with it, but people who've already built things might get annoyed by that.

apella12, 2024-02-14:
Before I add the PR can you see if this works.  (first time using the "compat" section.)
[zl7432.json](https://github.com/zwave-js/node-zwave-js/files/14281840/zl7432.json)

TheKorn2, 2024-02-14:
@apella12 That fixes the issue entirely!

PR #6675 (2024-04-02): fix(config): preserve root endpoint in Vision ZL7432
Fixes #6674

As this was not a new device config, I left EP1 to not disrupt current device owners.

PR description here
closed 2024-04-02 by AlCalzone