zwave-js/zwave-js #6680: Heltun HE-ZW-THERM-FL2: Override supported scales

AlCalzone, 2024-02-19:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/6666

<div type='discussions-op-text'>

<sup>Originally posted by **@steve-hodgkiss** February  8, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
Heltun thermostat is causing "unsupported scale or corrupted data" errors in log file. Other values in CC report such as power are being updated fine. 

What did you expect to happen instead?
Illuminance values should be updated to correct light level

Steps to reproduce the behavior:
  1. Go to 'node'
  2. Click on 're-interview'
  3. Scroll down to 'multilevel sensor'
  4. After re-interview after power cycle, illuminance is shown but never updated. Additional re-interview causes illuminance to disappear from sensor. In either case, illuminance value never gets updated.

Log file shows entries for illuminance and multiple recurring log messages
```
DRIVER   dropping CC with invalid values (Reason: Unsupported scale Lux or corrupted da
                                  ta)
```

Anything else we should know?

Only applies to this model of Heltun Thermostat. More recent model HE-HT01 does not suffer from this issue.

### Software versions

Z-Wave JS UI: v9.8..3

Home Assistant Z-Wave JS Addon: Add-on version: 3.3.2

### Device information

Manufacturer: Heltun
Model name: HE-ZW-THERM-FL2
Node ID: 062

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [X] The log includes a re-interview of the problematic device _(if applicable)_.

- [X] The log includes the problematic interaction with the device _(if applicable)_.

- [X] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/14210361/zwavejs_current.log)
</div>

AlCalzone, 2024-02-19:
The device reports support for the percentage scale, not Lux:
```
2024-02-08T12:37:24.612Z SERIAL « 0x010c0004003e0431060301cf0037                                      (14 bytes)
2024-02-08T12:37:24.614Z SERIAL » [ACK]                                                                   (0x06)
2024-02-08T12:37:24.616Z CNTRLR   [Node 062] [+] [Multilevel Sensor] supportedScales[3]: [Endpoint 0] [internal]
                                   0
2024-02-08T12:37:24.617Z DRIVER « [Node 062] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCSupportedScaleReport]
                                      sensor type:      Illuminance
                                      supported scales:
                                      · Percentage value
```
I'm assuming this is an off-by-one error in the support bitmask. Here, `0x01` means that scale 0 (percentage) is supported. For scale 1 (Lux), the value should be `0x02`.

We should override this with a compat flag

AlCalzone, 2024-03-09:
That sounds different. Please open another request for investigation (under discussions) and share all the information requested there.

maxime-scher, 2024-03-09:
Thanks for your answer. I opened it : [#6692]

PR #6711 (2024-04-02): fix(config): override supported sensor scales for HELTUN HE-ZW-THERM-FL2
fixes: #6680
closed 2024-04-02 by zwave-js-bot