zwave-js/zwave-js #6690: Zwave-JS-UI rejects OTA firmware file that Simplicity Studio is happy to program

TheKorn2, 2024-03-04:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Snap

### Z-Wave JS UI version

zwave-js-ui: 9.9.0.09b107a

### ZwaveJS version

zwave-js: 12.4.4

### Describe the bug

AEON labs ZW095 Gen5 HEM...  OTZ update file supplied directly from Aeotec is rejected with "Unable to extract firmware from file: No EOF record at end of file" message.

Load OTZ file in Simplicity Studio and it's more than happy to load it and program the end device with it.

### To Reproduce

Select a node (preferably a zw095, but should be trigger-able by just about anything). Go to advanced, firmware update, begin. Select attached OTZ file (which has been zipped for upload to github, but should be unzipped before attempting). 

See unhappy message.
[image]

### Expected behavior

[ZW095_HEM_G5_2P_200A_US_V1_33.zip](https://github.com/zwave-js/zwave-js-ui/files/14456156/ZW095_HEM_G5_2P_200A_US_V1_33.zip)

OTA update should proceed.

apella12, 2024-03-05:
On an unrelated note, your screen shot shows FW 1.37 which is the latest (I believe) and your file appears to be for FW1.33.  Are you sure you want to do this? ;-)

TheKorn2, 2024-03-05:
I have more than one. One is limited to 1.33 max, the other can handle 1.37. (I already upgraded both with SS.)

duylong, 2024-03-05:
I had a similar problem when I did my Aeotec updates. It seems to me that this is a problem with the otz file provided. Depending on the module, some files pass, others do not. I finally went through their EXE to do the update (Zensys.*).

robertsLando, 2024-03-05:
Zwave-js expects OTZ to be in Intel HEX format

TheKorn2, 2024-03-05:
@duylong The funny thing is the 1.37 update goes through just fine, but the 1.32 and 1.33 updates get kicked back. I wouldn't expect they'd change formats between those, but I guess anything is possible. (I have all of these if they'd help.)

robertsLando, 2024-03-06:
The only one that can tell you something about this is @AlCalzone

AlCalzone, 2024-03-06:
I'll look at the files when I find the time. If you can share both (one working, one not), that's definitely helpful.

TheKorn2, 2024-03-06:
@AlCalzone And with a waive of the magic wand...  This one updates just fine with zwave-js. (previous one is the one that gets kicked out.)  
[ZW095_HEM_G5_2P_200A_US_V1_37.zip](https://github.com/zwave-js/node-zwave-js/files/14509178/ZW095_HEM_G5_2P_200A_US_V1_37.zip)

TheKorn2, 2024-03-06:
This one also gets rejected.  
[ZW095_HEM_G5_2P_200A_US_V1_32.zip](https://github.com/zwave-js/node-zwave-js/files/14509183/ZW095_HEM_G5_2P_200A_US_V1_32.zip)

AlCalzone, 2024-04-08:
Yeah, I don't think it's worth going through the trouble to work around this.

@aeotec-ccheng FYI, this seems like an easy fix on your side.

closed 2024-04-08 by AlCalzone

Aeotec-ccheng, 2024-04-08:
I'll forward it up the teams - thanks @kpine @AlCalzone .