zwave-js/zwave-js #6693: IBT4ZWAVE Not working with zwave-js-ui: 9.9.0 and zwave-js: 12.4.4

Dites33, 2024-03-11:
[debuginfo.txt](https://github.com/zwave-js/zwave-js-ui/files/14550429/debuginfo.txt)
**zwave-js-ui: 9.9.0
zwave-js: 12.4.4
home id: 3900514980
home hex: 0xe87d22a4**

I finally managed to pair my ZWave BIDI interface through Nice to my Aotec key. Unfortunately, while the node responds perfectly to all commands, the portal does not move an inch (opening closing or anything else :))

Can you help me see things clearly?

_when closing_

`2024-03-10T13:50:02.547Z CNTRLR   [Node 150] failed to decode the message, retrying with SPAN extension...
2024-03-10 14:50:02.671 INFO Z-WAVE: [Node 150] Value updated: 38-0-targetValue null => 0
2024-03-10 14:50:02.674 INFO Z-WAVE: [Node 150] Value updated: 38-0-currentValue null => 0`

_when opening_

`2024-03-10T13:53:32.898Z` CNTRLR   [Node 150] failed to decode the message, retrying with SPAN extension...
2024-03-10 14:53:33.019 INFO Z-WAVE: [Node 150] Value updated: 38-0-targetValue null => 99
2024-03-10 14:53:33.020 INFO Z-WAVE: [Node 150] Value updated: 38-0-currentValue null => 99
2024-03-10T13:53:33.101Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
2024-03-10 14:53:38.157 INFO Z-WAVE: [Node 150] Value updated: 38-0-targetValue 99 => null
2024-03-10 14:53:38.158 INFO Z-WAVE: [Node 150] Value updated: 38-0-duration unknown => unknown
2024-03-10 14:53:38.159 INFO Z-WAVE: [Node 150] Value updated: 38-0-currentValue 99 => null`

_when stopping_

`2024-03-10T13:51:47.657Z CNTRLR   [Node 150] failed to decode the message, retrying with SPAN extension...
2024-03-10 14:51:47.779 INFO Z-WAVE: [Node 150] Value updated: 38-0-Open false => false`

[debuginfo.txt](https://github.com/zwave-js/zwave-js-ui/files/14550425/debuginfo.txt)

Dites33, 2024-03-19:
Hi @AlCalzone , anything i can do to help ?
Any logs i could retrieve somewhere for you ?
Just ask :)

robertsLando, 2024-03-28:
@Dites33 still wait a bit as @AlCalzone comes back from his parental leave see https://github.com/zwave-js/node-zwave-js/issues/6521

AlCalzone, 2024-04-09:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

That log should include a re-interview and you trying to control the device.

closed 2024-04-20 by zwave-js-assistant[bot]