zwave-js/zwave-js #6694: Missing device configuration: fibaro and fdg-212 and dimming don't work

Frank18000, 2024-03-11:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

fibaro  fdg212

### Manufacturer ID

0x010f

### Product Type

 0x0102

### Product ID

0x1000

### Firmware Version

v3.3

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4507

### Do you have a manual?

[FGD-212-EN-T-v1.3.pdf](https://github.com/zwave-js/node-zwave-js/files/14563269/FGD-212-EN-T-v1.3.pdf)

apella12, 2024-03-13:
This device is in the DB. Does your light support dimming? On quick look default of parameter 32 looks like there is an autocalibration.  You could try "0" to force dimming if you are sure your light supports it.

Frank18000, 2024-03-14:
I am migrating from domoticz to HA. The light always worked in domoticz so i don't think there is an electrical problem.
When i look under configuration i get this:

[image]

So i think somthing goes wrong with the indentification of the dimmer.

AlCalzone, 2024-03-14:
@Frank18000 please fill out https://github.com/zwave-js/node-zwave-js/discussions/new?category=request-support-investigate-issue and provide all the requested information, especially the interview log. We can investigate there.

closed 2024-03-14 by AlCalzone