zwave-js/zwave-js #6709: Issues with firmware upgrade - zwavejs need to be restarted between bootloader and firmware

richardstone, 2024-03-29:
I'm trying to upgrade the firmware on one of many devices connected to a single controller.
Devices I'm using:
- **Controller**: Dusun Gateway with 700 series chip
- **Device I'm trying to upgrade**: Haseman Electric RS-10PM2 with Silicon Labs ZGM130S - 700 series chip

**Upgrade steps:**
- select the device I want to upgrade
- press "Advanced"
- press "Begin" below "Firmware update"
- select bootloader for the first row (for target "0")
- select firmware for the second row (for target "1")
- press "Ok" to start the process.
[image]

After starting the upgrade process, the bootloader is uploaded to the device, then the device restarts when the bootloader has finished uploading. After 7 seconds the controller tries to continue with the second file, but the upload of that is never started. At this point I need to restart the docker container where zwavejs is running, wait until it checks and reconnects to all devices, then repeat the upgrade steps mentioned above, except with only selecting the second file. After this, the upgrade goes well, and at the end, the upgraded device gets reinterviewed, and the upgrade is finished successfully.

I have attached a stipped version of the logs. Sometimes a few packages are dropped during the upload, and in some cases I get "Failed to execute controller command" messages, but eventually the upgrade can succeed.

**Things I have tried:**
- "abort" the upload after the upload of the bootloader, but I got an error telling me that the firmware upgrade succeeded, so there is nothing to abort.
- upload the bootloader only, but with that, the upgrade could not be started.

When pinging the device after uploading the bootloader I see this in the logs:
`17:27:16.775 CNTRLR « [Node 021] ping successful
`

When triple click the zwave button on the device after bootloader is uploaded I get this in the logs:
`17:26:40.197 CNTRLR « [Node 021] Received updated node info
`

Logs of first upgrade attemt with both files selected: [zwavejs-upgrade-issue-log.txt](https://github.com/zwave-js/zwave-js-ui/files/14802137/zwavejs-upgrade-issue-log.txt)

Logs of second upgrade after restarting the controller and only issueing the firmware (for target "1"): [second-upgrade-fw-only-log.txt](https://github.com/zwave-js/zwave-js-ui/files/14802150/second-upgrade-fw-only-log.txt)

**Do you have any suggestions what I could try so that I don't need to restart the docker container in between the upload of the files? Is this a bug, or am I doing something wrong?**

richardstone, 2024-03-29:
I forgot to mention that I’m using zwavejsui 9.9.1

richardstone, 2024-04-02:
@robertsLando @AlCalzone let me know if I can further assist you with anything with this issue.
Thanks!

AlCalzone, 2024-04-02:
I came back from parental leave today and started digging through the heap of open issues and PRs. Might take a few days before I get to this.

richardstone, 2024-04-03:
Sure, take your time! :)

One addition to the issue is that the upgrade only stops after the bootloader if the device is connected to the controller with security. If there is no security, then the upgrade goes well through both bootloader and firmware.

AlCalzone, 2024-04-03:
Any chance you can make a driver log of that on loglevel "debug"? The ones above are on a higher level, so they don't contain the details I need to investigate.

AlCalzone, 2025-03-11:
Can you check if this is fixed for you in version 14.3.10 or newer?
https://github.com/zwave-js/zwave-js/blob/master/CHANGELOG.md#14310-2025-02-27

AlCalzone, 2025-05-09:
This should be fixed. If not, please open a new issue.

closed 2025-05-09 by AlCalzone