zwave-js/zwave-js #6755: Provisioning Entries Smart Start disappeared after upgrade to 9.10.2

rohankapoorcom, 2024-04-10:
### Checklist

- [x]  I am not using Home Assistant. Or: a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is not described there.
- [x] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.10.2.68fdf4d

### ZwaveJS version

12.5.2

### Describe the bug

After upgrading Z-Wave JS UI to 9.10.2.68fdf4d, the "Provisioning Entries" page has no entries when previously there were ~40 or so.

### To Reproduce

1. Have provisioned entries
2. Upgrade to 9.10.2.68fdf4d
3. Observe that the Provisioning Entries page shows no entries

### Expected behavior

The Provisioning Entries page should show the entries that were present before the upgrade.

### Additional context

Rolling back to the previously running version (and several versions behind) did **not** restore the entries.
zwave-js-ui: 9.9.1.f6dff21
zwave-js: 12.4.4

AlCalzone, 2024-04-10:
Everyone: Which versions did you upgrade **from**?

@rohankapoorcom the file should be `xxxxxxxx.jsonl` where the x is your home ID as hexadecimal.

rohankapoorcom, 2024-04-10:
@AlCalzone - I upgraded from zwave-js-ui: 9.9.1.f6dff21
zwave-js: 12.4.4

Unfortunately it seems my backup saving script failed, so I do not have a backup file to restore. Is it helpful if I send you that file (privately)?

AlCalzone, 2024-04-10:
I'll try to reproduce myself first. Will let you know if I need anything else.

AlCalzone, 2024-04-10:
Ok I can't reproduce. Can you send me the file via E-mail?

drowl87, 2024-04-10:
I'm coming here from ticket #3658, but I had the same issue as well. In looking at the xxxxxxxx.jsonl file, I see the various devices in there along with their respective dsk values, but the GUI for the Provisioning Entities didn't show any devices after the latest update. There was only one device that was in the provisioning state and that one totally disappeared when this issue was emerged since it had never been interviewed or accepted as a new node yet. I honestly am not sure if I was coming from 9.10.0 or 9.9.1 since I am just using the :latest tag. I had been running backups as well, so I do have some prior config files zipped/saved, but never ended up trying to do a restore. My hardware is a Synology NAS and I'm running HA and zwave-js in the "Container Manager" (i.e. Docker). I haven't changed my compose.yaml in a long time, so I don't think that is the issue, but here it is just in case. I'm happy to email over any files...just let me know where to send them if any of the devs want them. Thanks!

version: '3'
services:
  zwave-js-ui:
    container_name: zwave-js-ui
    image: zwavejs/zwave-js-ui:latest
    restart: unless-stopped
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    volumes:
      - /volume1/docker/zwave-js-ui:/usr/src/app/store
    ports:
      - '8091:8091'
      - '3000:3000'

AlCalzone, 2024-04-10:
@drowl87 can you send me your jsonl file aswell? I wonder if this is just the UI choking on some missing fields.

drowl87, 2024-04-10:
> @drowl87 can you send me your jsonl file aswell? I wonder if this is just the UI choking on some missing fields.

Just sent it via email!

robertsLando, 2024-04-10:
Could someone check the chrome console when you open the provisioning entries table? That may be an UI error in that case as something changed when we added the long range support

drowl87, 2024-04-10:
[image]
Is this helpful? Not too familiar with the Chrome dev area.

robertsLando, 2024-04-10:
@drowl87 Yeah that seems to show no errors on UI side, at least those errors seems to be related to some of your chrome extensions

rohankapoorcom, 2024-04-10:
> Ok I can't reproduce. Can you send me the file via E-mail?

Sent to the email on you Github profile.

I'm also not seeing any errors in the chrome console.

AlCalzone, 2024-04-12:
Confirmed this to be a bug in the driver. The data is still there, but it's not being exposed.

closed 2024-04-12 by zwave-js-bot

rohankapoorcom, 2024-04-12:
@AlCalzone - thanks for pushing out a fix. I just updated to the latest version of Z-Wave JS UI and am still seeing the same issue.

zwave-js-ui: 9.10.3.f2b51e2
zwave-js: 12.5.4

AlCalzone, 2024-04-12:
I think your file actually didn't have any provisioning entries ☹️
@drowl87's file did.

rohankapoorcom, 2024-04-12:
Sounds like something in the update must have clobbered them then. Pity my backup script broke.

Any way to regenerate them from an existing network without scanning all of the QR codes again?

AlCalzone, 2024-04-12:
It should be mostly possible with a bit of fiddling.
The DSK and **granted** security classes are in the cache file. **Requested** security classes are missing, but unless you want to change the security classes, setting them to be the same as the granted classes should work.
Supported protocols are missing, so we don't know if you could include one of those devices via Z-Wave Long Range.
User-provided device name and location should be in the `<homeid>.values.jsonl`, device fingerprint too.

kris-anderson, 2024-04-12:
I found this thread today when I found out all my Smart Start entries had disappeared.

I saw there was an update, and applied it. I'm now running version:
[v9.10.3](https://github.com/zwave-js/zwave-js-ui/releases/tag/v9.10.3) (2024-04-12)

But all my smart start devices are still missing 😭

alvinchen1, 2024-04-22:
Anyone still having this issue since the release of Release 9.11.1?  Rolled back to 9.9.1 to resolve the issue temporarily, but based upon the reports, not sure I want to update yet.  Thanks!

drowl87, 2024-04-22:
@alvinchen1 Yes I rolled back to 9.9.1, restored a backup, then successfully upgraded to 9.11.1 and my Smart Start entries remained.

alvinchen1, 2024-04-22:
@drowl87 , Thanks! Appreciate you taking the plunge for me.

lukasz-karolewski, 2024-04-25:
@alvinchen1 i'm on 9.11.1 Provisioning Entries are still missing. is the revert, restore, upgrade required to get them back?

AlCalzone, 2024-04-25:
You need the old jsonl file back when they are missing.

lukasz-karolewski, 2024-04-26:
this might be a good opportunity to make backup enabled by default for everyone ... I would have appreciated it now ...