zwave-js/zwave-js #6772: Best message in case of failure for association with different security level

duylong, 2024-04-16:
Hi,

I recently tried to add an association between 2 devices that did not have the same level of security. I got the errors below:

[image]

[image]

Once you know the problem it's easy, but if you don't, the message doesn't help.

Is it possible to improve the error message?

AlCalzone, 2024-06-16:
The driver will include the reason for the failure in the thrown error starting with version 13.

robertsLando, 2024-06-17:
Nice! No changes required on my side so as I already show the error thrown :)

AlCalzone, 2024-06-17:
do you use `isAssociationAllowed`?

robertsLando, 2024-06-17:
Yes:

```ts
for (const a of associations) {
					if (
						this._driver.controller.isAssociationAllowed(
							source,
							groupId,
							a,
						)
					) {
						this.logNode(
							zwaveNode,
							'info',
							`Adding Node ${a.nodeId} to Group ${groupId} of ${sourceMsg}`,
						)

						await this._driver.controller.addAssociations(
							source,
							groupId,
							[a],
						)

						return true
					} else {
						this.logNode(
							zwaveNode,
							'warn',
							`Unable to add Node ${a.nodeId} to Group ${groupId} of ${sourceMsg}, association not allowed`,
						)
					}
				}
```

AlCalzone, 2024-06-17:
Then you'll need to change something, see linked PR. I'll also write migration docs for v13 like usual.

commit ceff8521 (2024-07-05): refactor: replace `Controller.isAssociationAllowed` with `checkAssociation` (#6935)

fixes: #6772
closed 2024-07-05 by AlCalzone