zwave-js/zwave-js #6775: NVM restore: detect incorrectly formatted application version on 800 series

AlCalzone, 2024-04-18:
The 800 series application version NVM file `0x41000` is formatted differently on the Zooz stick than on the firmwares I can flash onto the devkit:

```
Zooz:   00 03 13 07
Devkit: 03 13 07 00
        ┌─ ┌─ ┌─ ┌─
        │  │  │  ??
        │  │  └ MAJOR
        │  └ MINOR
        └ PATCH
```

Since this is stored as little endian, the actual values used in code are `0x07130300` for the Zooz and `0x00071303` for the "official" firmware. When the migration routine for 7.20.0 -> 7.21.0 compares that value with `APP_VERSION_7_21_0` (`0x00071500`), the app version on the Zooz is bigger, so the migration does not trigger.

We should detect this situation, fix the application version in NVM, and not rely on the firmware built-in migration in this case.

closed 2024-04-23 by zwave-js-bot

danielsza, 2024-05-03:
This seems like the exact issue that I experienced. I purchased a 800 series ethernet adapter https://www.ebay.com/itm/166619186789 I restored my NVM from a Zooz 800 stick and the RF region became corrupt. I was not able to fix the corruption (z/ip, firmware update, NVM restore, reset the NVM) I ended up returning it as a result. 
I believe the seller had to flash it with a tool that I didn't own.

I ended up switching to Razberry Pro 7, along with a Tubesz Z-Wave PoE Kit. I restored the same NVM on this and it didn't corrupt the RF region .

AlCalzone, 2024-05-03:
I wonder if the Razberry migration routine works differently and isn't affected by this.
Z-Wave.me are running pretty customized firmware.