zwave-js/zwave-js #6776: Walli Controller FGWCEU-201 Value update failed

kilian-f-za, 2024-04-18:
Hallo, ich habe seit gestern einen Walli Controller FGWCEU-201, die Einbindung hat funktioniert aber die Configuration v4 funktioniert nicht. 
Die Parameter werden nicht übernommen und es kommt die Fehlermeldung Value update failed. Ich kann kein Parameter einstellen. 
Gibt es dazu einen Lösungsansatz oder kann ich noch was zu diesem Fehler beitragen? Ist ein zusätzliches Log hilfreich?

[image]

[image]

zwave-js-ui: 9.10.3
zwave-js: 12.5.4

kilian-f-za, 2024-04-19:
What can I do now or what needs to be changed?

robertsLando, 2024-04-19:
@kilian-f-za Wait for @AlCalzone to update the device configuration to disable supervision on that device as it doens't support it correctly

AlCalzone, 2024-04-19:
You can look up how we disabled supervision on other problematic devices and use a custom config file with the same applied.

Otherwise you'll have to wait for me to fix it on Monday, and then new releases of node-zwave-js and zwave-js-ui.

AlCalzone, 2024-04-22:
@kilian-f-za it looks like for parameter 150 we have an incorrect value size (1 byte instead of 2) in our config file. Parameters 13, 20, 152 and 153 should be correct.
Can you try setting them please? And please make a driver log of that.

kilian-f-za, 2024-04-22:
Yes, I can set the parameters 13, 20, 152 and 153. However, parameter 13 does not change the brightness on the device. When I press the current button, the switch lights up green.

It's Node 023
[zwavejs_2024-04-22.log](https://github.com/zwave-js/node-zwave-js/files/15062008/zwavejs_2024-04-22.log)

AlCalzone, 2024-04-22:
Ok those look good, so it's not the Supervision CC, but our incorrect config file. We can't do anything about the config parameters not behaving as they should though.

Side note: You don't have to remove and include the device to make a log, unless I ask you to :)

kilian-f-za, 2024-04-22:
What can I do now? So I can't set the parameters?

closed 2024-04-22 by zwave-js-bot

AlCalzone, 2024-04-22:
I'm fixing our config file, so you can set 150 and 151 as expected.
Setting the others appears to be working according to your log.

But if parameter 13 doesn't change the brightness even if the manual says so, that's nothing we can fix.

kilian-f-za, 2024-04-23:
When is the new release expected to be published?
I currently still have the right to return the Switch. 😄

AlCalzone, 2024-04-23:
Likely today, then Z-Wave JS UI will have to follow up with a release. And if you're using the Home Assistant addon, that will also need one.

If you can't wait, you can also test the file in advance. See https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files for an explanation how. Save [this](https://raw.githubusercontent.com/zwave-js/node-zwave-js/54cd24019942a89e0a166a524b253356a5a4f046/packages/config/config/devices/0x010f/fgwceu-201.json) as a `.json` file and put it into the folder mentioned in the documentation.

Don't forget to remove the custom file again after updating Z-Wave JS UI, otherwise you won't get any future fixes to that file.

PR #6779 (2024-04-22): fix(config): value size for Fibaro FGWCEU-201, params 150/151
fixes: #6776