zwave-js/zwave-js #6778: nvmedit: support ZWLR files

AlCalzone, 2024-04-19:


foxxcomm, 2024-05-07:
@AlCalzone --

Just tried NVM Backup of a ZAC93 controller and NVM restore to a second identical ZAC93 controller. Got no errors but the restored ZAC93 is missing all ZWLR nodes. Assume this is because ZWLR is not yet implemented in NVM backup/restore correct?

If not, let me know and I'll open an issue.

Thanks

AlCalzone, 2024-05-07:
That's likely the reason. You should be able to skip the compatibility check to get around this.

foxxcomm, 2024-05-07:
@AlCalzone --

Ok Thanks. When I see you merge  ZWLR NVM support I'll test again. No big deal at the moment as I only have some test nodes using ZWLR and they are performing wonderfully! Kudos!

Thanks

closed 2024-05-14 by zwave-js-bot