zwave-js/zwave-js #6781: Feature idea: attach desired config parameter values to SmartStart provisioning entries

AlCalzone, 2024-04-22:
This way, Z-Wave JS could configure devices immediately during the initial interview, and it would allow bulk provisioning and configuring in a more convenient way than having to script the post-interview configuration.

rjulius23, 2024-04-28:
Is it only possible via SmartStart ? Would it be possible to use REST API to bulk configure nodes? 

I give you a reason, you pair 100 devices, then you analyze teh network and you realize that there are some parameters with which you can improve the performance, right now oyu would need to click through all the devices to set the required values. If I could use REST API, I can write a simple python script that would go through the devices set the parameter sequentially. It would still take time, but i dont have to sit in front the machine :)

AlCalzone, 2024-04-29:
Depends on how you use Z-Wave JS. Z-Wave JS UI has a "driver function" feature for arbitrary scripting, which ships with a predefined snippet to copy parameters from one device to other identical ones.