zwave-js/zwave-js #6808: Configuration parameter No.117 allegedly missing from Shelly Qubino Wave 2PM (QNSW-002P16EU) (and maybe others?)

RaulDJ, 2024-05-04:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Shelly Qubino Wave 2PM

### Manufacturer ID

0x0460

### Product Type

0x0002

### Product ID

0x0081

### Firmware Version

v10.30.0

### Is the device listed on the Z-Wave Alliance website?

It isn't, here's the most similar product: https://products.z-wavealliance.org/products/4961?selectedFrequencyId=-1

### Do you have a manual?

Here's the device documentation: https://kb.shelly.cloud/knowledge-base/wave-2pm

Supposedly, there's a 1 byte parameter, "No. 117", that allows the user to reset the device when the apparently trigger happy protection algorithms happen to kick in, such in this case: https://kb.shelly.cloud/knowledge-base/wave-1pm#Wave1PM(EU)-ACmainsdisconnected .

That parameter seems to be common to a lot, if not all, Shelly Qubino devices, and it's further explained in some (but not all) of their manuals: https://products.z-wavealliance.org/ProductManual/File?folder=&filename=cert_portal/certs/1092/Extended_User_Guide_Wave_PlugUKV1.4.pdf

Maybe it's also missing on other configuration files from similar devices from this manufacturer.

Thanks a lot for your attention and the great work, keeping up on all these devices through years.

AlCalzone, 2024-05-06:
@QubinoHelp can you clarify which devices have this parameter?

RaulDJ, 2024-05-07:
@QubinoHelp 

Thanks a lot for the clarifications. That ability to remotely restore functionality is really needed at least for the "AC mains failure" alarm, as for what I've seen, it seems to happen fairly often with brownouts and such.

I understand that ignoring an overheat or overcurrent alarm without actually checking the physical condition of the devices attached to the relay box would be quite risky, but not being able to do it after a power loss (or a simple brownout) in areas with not great grid stability would be a deal breaker for people in a situation like mine, so I'll be eagerly waiting for those updates.

AlCalzone, 2024-05-07:
@QubinoHelp Thanks! I suppose we can add the parameter to our config files then with a `$if` condition to limit it to appropriate models/firmwares. Let me know if you need help with that.

QubinoHelp, 2024-05-07:
@AlCalzone That would be the case yes, we will add also the detached mode parameters that will have to be done with an if condition since they will be implemented in all of our devices but from a certain firmware up. 
I will need help when the time come :) thanks.

QubinoHelp, 2025-02-11:
There was some issues in the SDK that stop us to publish the updates, hopefully soon the SDK will be fixed enough so that we can continue to add the missing parameters, fixes and new features to our already released devices firmware.

RaulDJ, 2025-02-11:
@QubinoHelp Thanks for the heads up, let's hope for that to not take too long.

closed 2025-10-26 by RaulDJ