zwave-js/zwave-js #6811: Missing device configuration: <enter manufacturer and device name>

radhoo2k10, 2024-05-06:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Fibargroup

### Manufacturer ID

0x0304

### Product Type

0x0304

### Product ID

0x0304

### Firmware Version

v8.0.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4992?selectedFrequencyId=-1

### Do you have a manual?

https://www.elektroimportoren.no/docs/lib/95019-Brukerveiledning-5.pdf

[image]

[image]

[node_115.json](https://github.com/zwave-js/node-zwave-js/files/15224099/node_115.json)

AlCalzone, 2024-05-06:
@zwave-js-bot import config 4992 from zwa

closed 2024-05-24 by AlCalzone

radhoo2k10, 2024-05-30:
@AlCalzone Thanks for completing this. Got a question. I'm using zwave-js ui with mqtt and home assistant. 
Zwave-js ui just updated with zwave-js 12.9.1. But the controlls in Home assistant is still missing. Is it here I have to create an issue or is it Zwave-js ui? 

[image]

AlCalzone, 2024-05-30:
If you're connecting HA to Z-Wave JS UI via MQTT and not the more tightly coupled Z-Wave integration, I can't really help. 

The new shutter uses the Window Covering CC, which might not have the corresponding discovery support yet.

This might be an issue for Z-Wave JS UI then (which includes the MQTT discovery).

PR #6812 (2024-05-24): feat(config): add Fibaro FGR-224 Roller Shutter 4
fixes: #6811

This is a draft to add the above device. We appreciate any help cleaning it up and getting it into a releasable state. Please see https://github.com/zwave-js/node-zwave-js/issues/6811#issuecomment-2096942379 for instructions.