zwave-js/zwave-js #6837: Can't Change configuration parameters on Inovelli 2-1

Botched1, 2024-05-13:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.12.0.96eeb76

### ZwaveJS version

12.5.6

### Describe the bug

I have an Inovelli 2-1 device paired Z-Wave LR.

I noticed all of the configuration parameters listed looked incorrect. If I try to change any of them, though, I get an error and the old/incorrect value simply comes back.

When I try to change this:
[image]

What I see in the UI logs:

2024-05-11 09:22:19.492 DEBUG SOCKET: Event ZWAVE_API emitted to pol5kYzU8ynZr4dEAAAR
2024-05-11 09:22:19.493 INFO Z-WAVE: Calling api writeValue with args: [
  { nodeId: 258, commandClass: 112, endpoint: 0, property: 1 },
  25,
  null,
  [length]: 3
]
2024-05-11 09:22:19.493 INFO Z-WAVE: Writing 25 to 258-112-0-1
2024-05-11 09:22:19.559 ERROR Z-WAVE: Unable to write 25 on 112-0-1: Fail
2024-05-11 09:22:19.560 INFO Z-WAVE: Success zwave api call writeValue { status: 2 }
2024-05-11 09:22:20.660 INFO Z-WAVE: [Node 258] Value updated: 112-0-1 0 => 0

What I see in the zwave js logs:
2024-05-11T14:22:19.498Z DRIVER » [Node 258] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      196
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 82
                                    │ security class:  S2_Authenticated
                                    └─[SupervisionCCGet]
                                      │ session id:      52
                                      │ request updates: true
                                      └─[ConfigurationCCSet]
                                          parameter #:      1
                                          reset to default: false
                                          value size:       1
                                          value format:     UnsignedInteger
                                          value:            25
2024-05-11T14:22:19.504Z SERIAL « [ACK]                                                                   (0x06)
2024-05-11T14:22:19.506Z SERIAL « 0x010401a90152                                                       (6 bytes)
2024-05-11T14:22:19.507Z SERIAL » [ACK]                                                                   (0x06)
2024-05-11T14:22:19.508Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2024-05-11T14:22:19.523Z SERIAL « 0x011d00a9c400000100a67f7f7f7f03030600000000040100000ca10ca495bb    (31 bytes)
2024-05-11T14:22:19.524Z SERIAL » [ACK]                                                                   (0x06)
2024-05-11T14:22:19.524Z DRIVER « [REQ] [SendDataBridge]
2024-05-11T14:22:19.524Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:                           196
                                    transmit status:                       OK, took 10 ms
                                    routing attempts:                      1
                                    protocol & route speed:                Z-Wave Long Range, 100 kbit/s
                                    routing scheme:                        Resort to Direct
                                    ACK RSSI:                              -90 dBm
                                    ACK channel no.:                       3
                                    TX channel no.:                        3
                                    TX power:                              12 dBm
                                    measured noise floor:                  -95 dBm
                                    ACK TX power by destination:           12 dBm
                                    measured RSSI of ACK from destination: -92 dBm
                                    measured noise floor by destination:   -107 dBm
2024-05-11T14:22:19.556Z SERIAL « 0x011f00a80000010102119f03f200c175230da5ed5ab9a61d1baed900a5000e97e (33 bytes)
                                  f
2024-05-11T14:22:19.558Z SERIAL » [ACK]                                                                   (0x06)
2024-05-11T14:22:19.558Z DRIVER « [Node 258] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -91 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 242
                                    │ security class:  S2_Authenticated
                                    └─[SupervisionCCReport]
                                        session id:          52
                                        more updates follow: false
                                        status:              Fail
                                        duration:            0s                                                                                                                        

### To Reproduce

Pair Inovelli 2-1 device in Z-Wave LR mode.

Try o change ANY configuration parameter to any value from the GUI

### Expected behavior

I should be able to change the configuration values.

AlCalzone, 2024-05-13:
Unlikely, newer devices are specifically tested with Supervision.
@InovelliUSA any idea why the device (VZW31-SN) would not accept parameter changes? Was there a firmware update changing parameter sizes or so?

Botched1, 2024-05-13:
I can't answer that. But I will say I have tried it on a different hub that supports LR and I'm able to change the configuration values on that hub. 

Doesn't mean it can't be a device issue, just sharing another data point.

Botched1, 2024-05-13:
> Unlikely, newer devices are specifically tested with Supervision. @InovelliUSA any idea why the device (VZW31-SN) would not accept parameter changes? Was there a firmware update changing parameter sizes or so?

@InovelliUSA Just a side note, I see very similar behavior on other hubs when the device is paired in Z-Wave LR mode... So this is not Z-Wave JS specific.

mitchcapper, 2024-05-17:
I have this exact device and am able to set parameters without an issue. NOTE: I did have a problem with it failing to parse the QR code, so upgraded to master which added S2 long range, but in the process may have fixed this.  If you can you may want to try master.  I can confirm they can be set and do stick.

Botched1, 2024-05-17:
Is your device Laure in LR mode though? Mine works fine when paired NOT in LR mode.

It is only in LR mode it acts odd. And mine acts odd when in LR mode on 3 different hubs - not just Z-Wave js...

AlCalzone, 2024-05-17:
Did you solve your problem?

Botched1, 2024-05-17:
Sorry, no I did not. 

I assumed that since this appears to be a device firmware issue, and not a Z-Wave js issue, that the issue on here should be closed. 

If that is incorrect I can open it back up.

AlCalzone, 2024-05-17:
Good thinking. I'd keep it open still, because I want to investigate this more. Maybe there's a way to avoid this in the future by extending the certification tests.

Botched1, 2024-07-02:
In the end this appears to be a bad device. Somehow the memory is corrupted to the point that configuration parameters can not be changed paired in mesh or LR mode. Other devices of the same firmware version can be changed as expected.

This should be closed as a can't reproduce / device issue.

closed 2024-07-02 by Botched1