zwave-js/zwave-js #6838: Adding Second First Alert Smoke/CO Detector failure

tommyjlong, 2024-05-13:
This maybe a ZWaveJS question, but thought I would start here.
I successfully added a First Alert Smoke/CO detector using Smart Start using HA's Companion App (Settings->Device->Add Device->ZWave->Add Z-Wave device->)by scanning QR Code then putting device into inclusion mode and a few seconds later it added successfully.  I believe this inclusion was most likely directly with the controller (i.e. not NWI).

I tried adding a second First Alert Smoke/CO detector (same exact model number) using the same method as the first Smoke/CO detector, but I believe this second one was far away from the controller and thus this one was a NWI.  Anyway it failed to add.
In the ZWaveJS-UI->Smart Start-> Provisioning Entries there is a new entry for the second smoke detector with the correct DSK and says it is "Active". ZWaveJS-UI->status shows "inclusion stopped".

I moved the second Smoke/CO detector near the ZStick (Aeotec Gen5) and put the device into inclusion mode again, and still no luck with adding it.
I have tried factory reset the smoke detector, deleted the provisioning entry (using ZWaveJS UI trashcan on the entry), replugged ZStick, restarted ZWaveJS Addon, and this time used ZWaveJSUI to scan the QR code which recreated the provisioning entry for the second device, and when I next put the device into inclusion mode, still not successful.

About the only thing I see coming out of the ZWaveJS-UI logs that appears to correlate in time with putting the device into inclusion mode is the following (log level is silly) and this is repeatable :

```
....
2024-05-12T18:16:15.909Z SERIAL « 0x010b004986260514c5418ff4f3                                        (13 bytes)
2024-05-12T18:16:15.910Z SERIAL » [ACK]                                                                   (0x06)
2024-05-12T18:16:15.911Z DRIVER « [REQ] [ApplicationUpdateRequest]
                                    payload: 0x260514c5418ff4
2024-05-12T18:16:16.063Z SERIAL « 0x010b004986260514c5418ff4f3                                        (13 bytes)
2024-05-12T18:16:16.064Z SERIAL » [ACK]                                                                   (0x06)
2024-05-12T18:16:16.065Z DRIVER « [REQ] [ApplicationUpdateRequest]
                                    payload: 0x260514c5418ff4
2024-05-12T18:16:16.098Z SERIAL « 0x010b004986260514c5418ff4f3                                        (13 bytes)
2024-05-12T18:16:16.099Z SERIAL » [ACK]                                                                   (0x06)
2024-05-12T18:16:16.100Z DRIVER « [REQ] [ApplicationUpdateRequest]
                                    payload: 0x260514c5418ff4
2024-05-12T18:16:16.117Z SERIAL « 0x010b004986260514c5418ff4f3                                        (13 bytes)
2024-05-12T18:16:16.118Z SERIAL » [ACK]                                                                   (0x06)
2024-05-12T18:16:16.119Z DRIVER « [REQ] [ApplicationUpdateRequest]
                                    payload: 0x260514c5418ff4
2024-05-12T18:16:16.379Z SERIAL « 0x010b004986260514c5418ff4f3                                        (13 bytes)
2024-05-12T18:16:16.380Z SERIAL » [ACK]                                                                   (0x06)
2024-05-12T18:16:16.381Z DRIVER « [REQ] [ApplicationUpdateRequest]
                                    payload: 0x260514c5418ff4
2024-05-12T18:16:16.443Z SERIAL « 0x010b004986260514c5418ff4f3                                        (13 bytes)
2024-05-12T18:16:16.443Z SERIAL » [ACK]                                                                   (0x06)
2024-05-12T18:16:16.446Z DRIVER « [REQ] [ApplicationUpdateRequest]
                                    payload: 0x260514c5418ff4
....
```
If I should post this issue/question somewhere else, just let me know where.

AlCalzone, 2024-05-13:
You're not supposed to start the inclusion process manually when using SmartStart. The controller will listen for devices trying to join and the devices regularly send out inclusion requests.

If that doesn't happen, I've usually had luck after doing one or more of the following things:
- **Exclude** the device you're trying to join
- Factory reset the device (you already have)
- Power cycle the device if possible, e.g. by cutting power or shortly removing the battery.

Also, which firmware is your stick on? SmartStart was only added in [firmware 1.2](https://aeotec.freshdesk.com/support/solutions/articles/6000252294-z-stick-gen5-v1-02-firmware-update).

tommyjlong, 2024-05-13:
Any ideas what the `ApplicationUpdateRequest` is from the Logs?

Exclusion/PowerCycle - The exclusion process at the device itself is the same as inclusion which is: with the device's batteries pulled out, hold the device's "test" button down while inserting the batteries.  Yeah I've done this many times too.  If you mean to say do exclusion at the controller, since it was never added successfully and thus not in the UI, is there a way to exclude it at the controller?

ZStick firmware: I upgraded the Gen5 ZStick to 1.2 some time ago..... Smart Start has worked before with this stick/firmware, so hopefully it is not a Zstick problem.  Oh and I have around 25-30 devices known by the ZStick, most are the older dumb inclusion types.

Anyway, thanks a bunch for taking the time to look at this for me.  This is the first time I've never been able to include a device.

tommyjlong, 2024-05-13:
I was able to add this via manual non-secure mode. So I'll close this.

closed 2024-05-13 by tommyjlong

AlCalzone, 2024-05-14:
Good that you solved your issue - I'll still answer the questions.

> Any ideas what the `ApplicationUpdateRequest` is from the Logs?

These specific ones are of type `SmartStart_NodeInfo_Received`, which means an already included smart start node has been powered up. So the device still considered itself to be part of some network.

> If you mean to say do exclusion at the controller, since it was never added successfully and thus not in the UI, is there a way to exclude it at the controller?

Yeah, just do it as you normally would if the device was part of the current network. The device will receive the messages to leave its current network. You'll know that it works if Z-Wave JS tells you that the **exclusion was stopped without changes** before the timeout ends.