zwave-js/zwave-js #6878: Nodes stuck on ProtocolInfo after migration, even though ProtocolInfo is received?

proycon, 2024-05-30:
I'm migrating from openzwave to zwavejs. Some of my devices (mostly Fibaro FSM01) migrated fine, but many others are stuck on 'ProtocolInfo' for over a day now, with `Unknown manufacturer 0xXXXX`. Prior to the migration, these devices were also working fine on openzwave.

I forced an interview on one of the stuck nodes:

```
2024-05-30T09:47:24.918Z CNTRLR   [Node 014] Beginning interview - last completed stage: None                                                                                                                                                   
2024-05-30T09:47:24.918Z CNTRLR   [Node 014] new node, doing a full interview...                                                                                                                                                                
2024-05-30T09:47:24.919Z CNTRLR » [Node 014] querying protocol info...                                                                                                                                                                          
2024-05-30T09:47:24.922Z DRIVER   all queues busy                                                                                                                                                                                               
2024-05-30T09:47:24.924Z SERIAL » 0x010400410eb4                                                       (6 bytes)                                                                                                                                
2024-05-30T09:47:24.925Z DRIVER » [REQ] [GetNodeProtocolInfo]                                                                                                                                                                                   
                                    payload: 0x0e                                                                                                                                                                                               
2024-05-30T09:47:25.742Z SERIAL « [ACK]                                                                   (0x06)                                                                                                                                
2024-05-30T09:47:25.745Z SERIAL « 0x0109014113960101010634                                            (11 bytes)                                                                                                                                
2024-05-30T09:47:25.748Z SERIAL » [ACK]                                                                   (0x06)                                                                                                                                
2024-05-30T09:47:25.749Z DRIVER « [RES] [GetNodeProtocolInfo]                                                                                                                                                                                   
                                    payload: 0x139601010106                                                                                                                                                                                     
2024-05-30T09:47:25.768Z CNTRLR « [Node 014] received response for protocol info:                                                                                                                                                               
                                  basic device class:    Controller                                                                                                                                                                             
                                  generic device class:  Remote Controller                                                                                                                                                                      
                                  specific device class: Simple Remote Control                                                                                                                                                                  
                                  node type:             Controller                                                                                                                                                                             
                                  is always listening:   false                                                                                                                                                                                  
                                  is frequent listening: false                                                                                                                                                                                  
                                  can route messages:    false                                                                                                                                                                                  
                                  supports security:     false                                                                                                                                                                                  
                                  supports beaming:      true                                                                                                                                                                                   
                                  maximum data rate:     100000 kbps                                                                                                                                                                            
                                  protocol version:      3                                                                                                                                                                                      
2024-05-30T09:47:25.783Z CNTRLR   [Node 014] Interview stage completed: ProtocolInfo                                                                                                                                                            
2024-05-30T09:47:25.783Z CNTRLR » [Node 014] querying node info...            
```

This looks good to me, however, the webinterface still shows the device as stuck on ProtocolInfo (and I assume therefore it doesn't propagate over MQTT to HomeAssistant either).

Pressing buttons that device also triggers events in the log. So the device seems to work fine.

Am I overlooking something? Is this a bug?

Zwave-JS 12.5.6
Zwave-JS-UI 9.12.0.96eeb76
Running in Podman using the Docker image.

proycon, 2024-05-30:
> Does this return then? 

Good question.. unfortunately my logs have rotated again since I restarted the container many time so I lost that info now...

In the meantime I worked around this issue by re-including the devices that were stuck. Now they work (but I'm not sure why a reinclude was needed as it worked before the migration, but perhaps I don't understand enough of zwave and the inclusion sends some extra info that openzwave processed at the time and wasn't available for zwavejs until now).

AlCalzone, 2024-05-30:
The device is a sleeping device, so it has to be woken up for the interview to happen.

Interview stage = Protocol Info means that this stage was **last completed**. This is read from the controllers memory, not the device.
The next part requires communication with the device, which cannot happen while it is asleep.

AlCalzone, 2024-05-30:
Side note:
> propagate over MQTT to HomeAssistant 

You should use the Z-Wave integration, not MQTT. The former is more tightly coupled with Z-Wave JS.

proycon, 2024-05-30:
> Interview stage = Protocol Info means that this stage was **last
> completed**. This is read from the controllers memory, not the device.
> The next part requires communication with the device, which cannot
> happen while it is asleep.

Ah, thanks for the clarification. That makes sense. I had indeed assumed it came from the
device rather than the controller.

> You should use the Z-Wave integration, not MQTT. The former is more tightly coupled with Z-Wave JS.                                    

That's why I chose MQTT, I don't want the coupling to be too tight. I
have MQTT as the backbone for everything and want to have the
flexilibity to move things away from home assistant to simpler scripts. 
I also like that zwavejs-ui is a separate thing in its own container now.

AlCalzone, 2024-05-31:
By being coupled more tightly I meant in terms of functionality. You can still run Z-Wave JS UI separately, and even use the MQTT part.
However the mqtt discovery is limited, and not as actively maintained. So to get Z-Wave JS UI  connected to HA, I **strongly** recommend the websocket based integration.

closed 2024-06-16 by AlCalzone