zwave-js/zwave-js #6888: Cannot specify TCP connection as serial connection

elronzo, 2024-06-03:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is not described there.
- [ ] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

latest

### ZwaveJS version

latest

### Describe the bug

I cannot specify a TCP (tcp://somehost.lan:someport) connection to the zwave stick thus zwave-js-server supports it.

### To Reproduce

Try to use tcp://host:port instead of /dev/dev-uuid

### Expected behavior

Allow specifying a tcp connection.

### Additional context

As my zwave-js-ui instance runs in a k3s cluster I would highly appreciate the possibility to specify a TCP connection for the actual zwave device as I do not want to build a custom image including socat or ser2net.

elronzo, 2024-06-03:
Problem is that I have set a homeID in FHEM already and included several zwave nodes. In order to migrate zwave to zwave-js-ui I will probably need a way to set this particular homeId in zwave-js-ui...

AlCalzone, 2024-06-03:
Good luck then

closed 2024-06-03 by AlCalzone