zwave-js/zwave-js #6891: Allow Manual Region Specification

mwaterbu, 2024-06-05:
I noticed that I was not being offered an update which should have been available (for a Zooz ZSE43). I believe the issue is that I do not have a region specified by my ZWave stick, so the update cannot match.

In cases like this, could it be possible to manually specify a region (for older sticks)?

Also, just to check that my assumption is correct, where would I see this region, if my stick did allow retrieving it?

As a sidenote, zwave-js/firmware-updates#82 would be useful here, as at first glance, it simply looked like my device was up to date, when in reality, the update check could not be properly executed.

closed 2024-11-05 by zwave-js-bot

mwaterbu, 2025-02-18:
So I see this in the changelog for the Z-Wave JS HA Add-on (v 0.10.0):
> Z-Wave JS: Allow specifying RF region for OTA firmware updates if the region is unknown or cannot be queried

But I don't see a way to set this through the add-on configuration. Is it a hidden setting somewhere?

AlCalzone, 2025-02-18:
It should be right next to the OTA update controls, but only if your controller does not allow querying the region.

mwaterbu, 2025-02-18:
I don't see any OTA update controls. To be clear, I am using the Z-Wave JS HA add-on, not Z-Wave JS UI.

The logs list my controller as a legacy controller, which I thought means it cannot query the region:
```
CNTRLR   querying additional controller information...
CNTRLR   received additional controller information:
         Z-Wave API version:         5 (legacy)
         Z-Wave chip type:           ZW050x
         node type                   Controller
         controller role:            primary
         controller is the SIS:      true
         controller supports timers: false
```

AlCalzone, 2025-02-18:
HA hasn't added any UI to use this feature yet. It's only in Z-Wave JS UI right now.

PR #7369 (2024-11-05): feat: allow specifying RF region for OTA firmware updates if unknown
This PR expands the options for `Controller.getAvailableFirmwareUpdates` and adds a new `rfRegion` property that will be used as a fallback in case the controller's RF region cannot be determined.

This allows using the OTA update service with older controllers that have no support for querying or changing the RF region. This feature should be used with caution, as specifying the wrong region may result in bricking the device.

fixes: #6891
fixes: https://github.com/zwave-js/node-zwave-js/issues/7323