zwave-js/zwave-js #6893: `supportsLongRange` is not being set when `enableSoftReset` is set to `false`

AlCalzone, 2024-06-05:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/6872

<div type='discussions-op-text'>

<sup>Originally posted by **@marcus-j-davies** May 26, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

As discussed on Discord:
It appears that when `enableSoftReset` is set to `false`, it seems to not set a value for `supportsLongRange` whether or not this is cosmetic or an affect on the driver itself is all yours 😉 

See : Example video

https://github.com/zwave-js/node-zwave-js/assets/55892693/aa99848e-d471-49f5-92c6-92b7821aa65a

### Software versions

**Driver Version** : 12.9.1

### Device information

Controller:

**Manufacture**	AEON Labs
**Modal**	ZW090
**Firmware Version**	1.2

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [ ] The log includes a re-interview of the problematic device _(if applicable)_.

- [ ] The log includes the problematic interaction with the device _(if applicable)_.

- [ ] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

2 log files : 

 - When `enableSoftReset` is `false`
  - When `enableSoftReset` is `true`
  
[zwavejs_2024-05-26 - False.log](https://github.com/zwave-js/node-zwave-js/files/15448296/zwavejs_2024-05-26.-.False.log)

[zwavejs_2024-05-26 - True.log](https://github.com/zwave-js/node-zwave-js/files/15448300/zwavejs_2024-05-26.-.True.log)

</div>

marcus-j-davies, 2024-06-06:
PR : https://github.com/zwave-js/node-zwave-js/pull/6894

I can confirm `supportsLongRange` is now set regardless of **soft reset**

AlCalzone, 2024-06-06:
fixed by https://github.com/zwave-js/node-zwave-js/pull/6894