zwave-js/zwave-js #6902: Unable to update parameters for Minoston MP21ZD

webmogul1, 2024-06-06:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

PKG executable

### Z-Wave JS UI version

9.13.4

### ZwaveJS version

12.10.0

### Describe the bug

Trying to update configuration parameters 3, 5, 15, 21, 16, 20, 22, 23, 26 (Value Update Failed)

### To Reproduce

Just try to update configuration parameter 23

### Expected behavior

Parameter is updated

### Additional context

```
2024-06-05T15:09:09.636Z SERIAL » 0x011100a900010100039f013d250000000038fa                            (19 bytes)
2024-06-05T15:09:09.636Z DRIVER » [Node 256] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      56
                                  └─[Security2CCNonceGet]
                                      sequence number: 61
2024-06-05T15:09:09.640Z SERIAL « [ACK]                                                                   (0x06)
2024-06-05T15:09:09.642Z SERIAL « 0x010401a90152                                                       (6 bytes)
2024-06-05T15:09:09.642Z SERIAL » [ACK]                                                                   (0x06)
2024-06-05T15:09:09.643Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2024-06-05T15:09:09.657Z SERIAL « 0x011d00a93800000100a27f7f7f7f0303060000000004010000049e04a5967e    (31 bytes)
2024-06-05T15:09:09.658Z SERIAL » [ACK]                                                                   (0x06)
2024-06-05T15:09:09.658Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:                           56
                                    transmit status:                       OK, took 10 ms
                                    routing attempts:                      1
                                    protocol & route speed:                Z-Wave Long Range, 100 kbit/s
                                    routing scheme:                        Resort to Direct
                                    ACK RSSI:                              -94 dBm
                                    ACK channel no.:                       3
                                    TX channel no.:                        3
                                    TX power:                              4 dBm
                                    measured noise floor:                  -98 dBm
                                    ACK TX power by destination:           4 dBm
                                    measured RSSI of ACK from destination: -91 dBm
                                    measured noise floor by destination:   -106 dBm
2024-06-05T15:09:09.667Z SERIAL « 0x012200a80000010100149f025601dbe0ada122d98de21ed3e5c37d904b08009d0 (36 bytes)
                                  003984b
2024-06-05T15:09:09.668Z SERIAL » [ACK]                                                                   (0x06)
2024-06-05T15:09:09.668Z DRIVER « [Node 256] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -99 dBm
                                  └─[Security2CCNonceReport]
                                      sequence number:  86
                                      SOS:              true
                                      MOS:              false
[... 82 lines cut ...]
                                    ACK channel no.:                       3
                                    TX channel no.:                        3
                                    TX power:                              4 dBm
                                    measured noise floor:                  -96 dBm
                                    ACK TX power by destination:           4 dBm
                                    measured RSSI of ACK from destination: -91 dBm
                                    measured noise floor by destination:   -99 dBm
2024-06-05T15:09:10.757Z SERIAL « 0x012200a80000010100149f035800e71e519695fbb03feb7937d4f2055b08009f0 (36 bytes)
                                  0049ca8
2024-06-05T15:09:10.759Z SERIAL » [ACK]                                                                   (0x06)
2024-06-05T15:09:10.760Z CNTRLR   [Node 256] [~] [Configuration] 2: 0 => 0                          [Endpoint 0]
2024-06-05T15:09:10.761Z DRIVER « [Node 256] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -97 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 88
                                    │ security class:  S2_Authenticated
                                    └─[ConfigurationCCReport]
                                        parameter #: 2
                                        value size:  1
                                        value:       0
2024-06-05T15:09:11.773Z CNTRLR   [Node 256] Timed out while waiting for a response from the node (ZW0201)
2024-06-05T15:09:11.773Z DRIVER   all queues idle
2024-06-05T15:09:19.737Z DRIVER   destroying driver instance...
2024-06-05T15:09:19.752Z DRIVER   driver instance destroyed
```

AlCalzone, 2024-06-06:
This might be related to https://github.com/zwave-js/node-zwave-js/pull/6686
Can you confirm whether the manual for your device is the same as https://cdn-files.myshopline.com/file/store/1701829637050/MP21ZD(800-Series)-Manual.pdf ?
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of a re-interview, loglevel `debug` and attach it here as a file (drag & drop into the text field).

webmogul1, 2024-06-06:
I don't have the manual anymore.

After enabling the log and trying to re-interview the node I started getting the "failed to transmit" issue. Then the mess started and I got several dead nodes. 

I have not experienced the "failed to transmit" issue before. (Zooz 800 Stick).

Right now I am trying to get things back how they were.

webmogul1, 2024-06-07:
here you go.
[zwavejs_current.log](https://github.com/user-attachments/files/15741300/zwavejs_current.log)

Using Zooz 700 stick.

AlCalzone, 2024-06-10:
Thanks. This will be fixed by the linked PR in next release. Unfortunately, you'll have to re-interview once more afterwards.