zwave-js/zwave-js #6910: Zooz ZEN30 switch contains a duplicate switch and light entity

AlCalzone, 2024-06-07:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/6589

<div type='discussions-op-text'>

<sup>Originally posted by **@jaygorrell** January  6, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
Adding a Zooz ZEN30 switch results in both a duplicated light and switch entity.

[image]

What did you expect to happen instead?
I'd expect to see a single switch and a single light.

Steps to reproduce the behavior:
  1. Add a new ZEN30 to HA

Anything else we should know?
I have 3x ZEN30 devices and a handful of ZEN71 and ZEN72 devices. The ZEN30 is the only one that has this issue. After opening a thread on the forums [here](https://community.home-assistant.io/t/zooz-devices-with-duplicate-entities/666883), the following was noted:
> The ZEN30 entities are duplicated because Zooz made a poor design decision related to multi-channel endpoints. https://github.com/zwave-js/node-zwave-js/pull/4736
>
> However there should only be a [duplicate switch](https://github.com/zwave-js/node-zwave-js/issues/4724#issuecomment-1163179279) entity, not a second light one, so maybe they fixed the problem in this latest revision of the device and the workaround is not required. You could [submit an issue](https://github.com/zwave-js/node-zwave-js/issues/new/choose) to node-zwave-js, including the full driver debug logs of a re-interview, and/or disable the entities that aren’t useful to you (firmware changelog doesn’t indicate anything).

### Software versions

Driver (node-zwave-js): 12.3.0

Home Assistant Z-Wave Integration: 1.33.0

### Device information

Manufacturer: Zooz
Model name: ZEN30
Node ID: 21

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [X] The log includes a re-interview of the problematic device _(if applicable)_.

- [X] The log includes the problematic interaction with the device _(if applicable)_.

- [X] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

[home-assistant_zwave_js_2024-01-06T04-17-17.737Z.log](https://github.com/zwave-js/node-zwave-js/files/13848364/home-assistant_zwave_js_2024-01-06T04-17-17.737Z.log)
</div>

AlCalzone, 2024-06-07:
As investigated in the discussion, we no longer need to preserve the root endpoint for the 800 series version of this device. The endpoints are behaving as they should now.

closed 2024-06-14 by zwave-js-bot

AlCalzone, 2024-06-14:
Will be fixed next release, after a re-interview.

jaygorrell, 2024-06-14:
Nice, thank you @AlCalzone!

brettonw, 2024-06-19:
Confirmed in the wild. Thank you.

PR #6931 (2024-06-14): fix(config): remove endpoint workaround for Zooz ZEN30 800LR
fixes: #6910