zwave-js/zwave-js #6920: Roller shutter switch Basic parameter missing, existing one have it

HolBaum5, 2024-06-12:
Hello everyone,

I have several TechniSat roller shutter switches, all of which have the same FW and SDK version. 
Now I have added a new roller shutter switch with the same FW and SDK version. 
On the new switch, I no longer see the Basic parameter in which the current shutter position is stored. Can I add it manually somehow?

**Old roller shutter switch**
[image]

**New roller shutter switch**
[image]

AlCalzone, 2024-06-12:
What's in Multilevel Switch?

HolBaum5, 2024-06-12:
Okay I understand, I have re-interveied all roller shutters and switched to the parameter 

MultilevelSwitchCCReport current value

closed 2024-06-12 by HolBaum5