zwave-js/zwave-js #6937: Zwave js ui to reset the controller when it gets unresponsive

radhoo2k10, 2024-06-17:
**Is your feature request related to a problem? Please describe.**
The 700/800 controllers have a bug that silicons labs has not fixed for over a year. The controller becomes unresponsive and needs to reset to work again.

https://www.silabs.com/documents/public/release-notes/SRN14930-7.21.3.0.pdf?fbclid=IwZXh0bgNhZW0CMTAAAR2W0khiIJ51yKAr8wGIcBq62NY68o-RfjLfgqq2RvEvb3fX9JMtiHxaL-Q_aem_AV1vYKD7COsIx7VM3qIiSUAdzgIi5sa01Y-TAp14nCOjJUpUAqZlt0gVXyF1tWhHtO3BfL2irVHHUOr-OmU8qGN3#page8

[image]

**Describe the solution you'd like**
I want the zwave js ui to do what silicon labs recommends from the host:
"When the controller is locked replying with the status, TRANSMIT_COMPLETE_FAIL, the host should reset the controller. "
"This low occurrence issue can be mitigated by the host" <- this is a lie. I have to reset the aoetec 7 controller twice a day now.

**Describe alternatives you've considered**
None. As of now I restart the zwave js ui docker container when the controller is unresponsive.

AlCalzone, 2024-06-17:
It is: https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md#12101-2024-06-06

The alternative for older versions would be using the **Soft reset** function from the advanced menu that's already built into Z-Wave JS UI.

closed 2024-06-17 by AlCalzone

robertsLando, 2024-06-17:
Ok so just update to ZUI 9.14.0 that I just released

radhoo2k10, 2024-06-17:
Thanks for the help! I hope the WAF factor increases :)