zwave-js/zwave-js #6938: CT101 thermostat not updating values until 'Refresh' or re-interview is clicked in the UI manually

automaton82, 2024-06-17:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Snap

### Z-Wave JS UI version

9.12.0

### ZwaveJS version

12.9.1

### Describe the bug

For some reason with the latest version of z-wave JS UI, when my thermostat updates the operating state (such as when the AC or furnace come on), there is no update given to z-wave JS UI until either 'Refresh' is clicked, or a re-interview is done manually in the UI. The moment this is done, all values are updated properly.

### To Reproduce

Use the CT101 thermostat, join it in a network and then after it's running, change the temperature until it kicks on the furnace or AC. The value for 'Operating State v2' will not update even if you wait indefinitely, nor will the temperature.

### Expected behavior

This should update like before.

### Additional context

This has happened to me before with earlier versions but it always sorted itself out after re-interviewing. With the latest version nothing I do seems to fix it.

Here's an example log from today:

```
6/14/2024, 1:57:09 PM - VALUE UPDATED
Arg 0:
└─commandClassName: Thermostat Operating State
└─commandClass: 66
└─property: state
└─endpoint: 0
└─newValue: 2
└─prevValue: 0
└─propertyName: state
6/14/2024, 1:37:39 PM - INTERVIEW COMPLETED
6/14/2024, 1:37:39 PM - READY
6/14/2024, 1:37:39 PM - INTERVIEW STAGE COMPLETED
Arg 0:
Complete
6/14/2024, 1:37:39 PM - INTERVIEW STAGE COMPLETED
Arg 0:
OverwriteConfig
```

Notice I did a manual re-interview at 1:37pm. At this point, `Operating State` was `Idle`. I then kicked on the AC and let it run for about 10 minutes, but z-wave JS UI did not register the change.

I then manually clicked `Refresh` in the UI which caused the update at 1:57pm and showed the correct value, `Cooling`.

automaton82, 2024-06-19:
I think you're right, I think it's a CT101 device issue. I think it doesn't push out value updates and must be polled.

Indianb0y016, 2024-06-19:
I can chime in as I have a Ct100 and used to use a CT101.
I can confirm that I recently began having issues where the device would send out its temperature values on every change. The fan state would send pretty regularly, with a few times being missed here and there. 
However, as of late, none of the parameters are receiving updates at all, even with various troubleshooting steps applied. In fact, I am actually running into some other issues with other nodes not having full node data points, like my zooz q sensors.
I am investigating a little further and am gonna create a separate issue if needed, but I can assure that the CT100 SHOULD send out temperature updates regularly, including fan state (usually); assuming the associations are configured correctly.

PeteRager, 2024-06-20:
Maybe unrelated but I am seeing problems with polling in the latest release, in that after a period of time, polls just get dropped.  I'm debugging to see which version the problem starts occurring in.

Hilo4321, 2024-10-28:
Thought I'd toss a log in here since I was fixing something else, and I see this behavior also. 

I'd call out that this does update sometimes, but generally I don't rely on it because it is so sporadic.

[zwavejs_current.log](https://github.com/user-attachments/files/17548622/zwavejs_current.log)

Hilo4321, 2024-11-01:
@robertsLando sorry for the ping, but I'm not sure how to reopen this. Seems like I don't have permission.

robertsLando, 2024-11-04:
cc @AlCalzone

mikedm139, 2026-01-30:
@AlCalzone Thanks for having a look. The thermostat seems to work fine. I can control it via my Home Assistant instance integrated via Zwave-js-ui. However, it doesn't automatically update the Operating State (CC 66). I know it used to do so although I don't know at what point it stopped doing so. It only occurred to me recently since I was going to create an automation that involved tracking the HVAC time in an active state.

automaton82, 2026-01-31:
Yes, I have also always had this (thus the bug entry). I agree at this point I think it's the device, not zwavejs.

To 'fix' it, I have an automation running every few minutes with:

```
data:
  entity_id:
    - climate.z_wave_thermostat_3
  refresh_all_values: true
action: zwave_js.refresh_value
```

It's been working fine for years.