zwave-js/zwave-js #6960: Missing device configuration: <Sensative Strips Guard 800 >

czdanika, 2024-06-24:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Sensative Strips Guard 800

### Manufacturer ID

0x019a

### Product Type

0x0700

### Product ID

0004.0004

### Firmware Version

36.27

### Do you have a manual?

[Sensative Strips Guard 800
](https://manual.zwave.eu/backend/make.php?lang=en&sku=SENEGUARD800&cert=)

AlCalzone, 2024-06-25:
> ### Manufacturer ID
> 
> 0x019a
> ### Product Type
> 
> 0x0700
> ### Product ID
> 
> 0004.0004

Where did you get those? 0004.0004 is not a format that's exposed in Z-Wave JS UI or Home Assistant.
Please report what's shown in the UI, so we can be sure to have the right ID.

AlCalzone, 2024-06-25:
According to the dump it is recognized already, so no need to add it.
```
  "manufacturer": "Sensative AB",
  "label": "11-01-022",
  "description": "Strips Guard 700",
  "fingerprint": {
    "manufacturerId": "0x019a",
    "productType": "0x0004",
    "productId": "0x0004",
    "firmwareVersion": "56.41.0",
    "hardwareVersion": 3
  },
  "interviewStage": "Complete",
  "ready": true,
```

To understand why it doesn't work with the other stick, please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) on loglevel `debug`. While making the log, please try including it again. Afterwards attach the log here as a file (drag & drop into the text field).

czdanika, 2024-06-25:
Hello!

After today's  JS updates, i tried to include it, now it recognized the device. But I made the log, please find attached. 
[zwave-js-ui-store.zip](https://github.com/user-attachments/files/15973905/zwave-js-ui-store.zip)

AlCalzone, 2024-06-25:
Sounds like another case of the bug fixed in https://github.com/zwave-js/node-zwave-js/releases/tag/v12.11.2

closed 2024-06-25 by AlCalzone