zwave-js/zwave-js #6983: Missing device configuration: Nortek GoControl GD00z-8-GC

jakew02, 2024-07-06:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Nortek GoControl GZ00Z-8-GC

### Manufacturer ID

0x014F

### Product Type

0x4744

### Product ID

0x3531

### Firmware Version

3.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3710

### Do you have a manual?

[10025383 Rev-B - GD00Z-8 GC_WEB.pdf](https://github.com/user-attachments/files/16114596/10025383.Rev-B.-.GD00Z-8.GC_WEB.pdf)

jakew02, 2024-07-06:
This device is currently showing as a GD00Z-7 on my setup.

AlCalzone, 2024-07-09:
We know of 4 different entries of the Z-Wave Alliance website, for the GD00Z-6, -7 and -8. All of them use the same IDs it seems. The entry for the -6 also mentions firmware 3.0, so I have really no clue how to distinguish them: https://products.z-wavealliance.org/products/2904

I guess I'll just mention all of them in the label.

closed 2024-07-09 by zwave-js-bot