zwave-js/zwave-js #6985: FGRM222 does not advertise support for Multilevel Switch CC

AlCalzone, 2024-07-09:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/6976

<div type='discussions-op-text'>

<sup>Originally posted by **@Patator33** July  4, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
Since 12.12.1, the FGRM222 (v22.22) have no multilevel switch. So i can just open and close with a binary switch.
When i use the wall switch manually, a multlilevel Switch v0 appear with just a current value.
With a FGR222, i have a multilevel switch v3 with possible action like target value.

What did you expect to happen instead?
have a complete multilevel swtich with positionning

Steps to reproduce the behavior:
re-interview node wiil do nothing but the multilevel swtch v0 desappear.

Anything else we should know?

### Software versions

zwave-js-ui: 9.14.4
zwave-js: 12.12.1
home id: 3527579196
home hex: 0xd242963c
install on LXC promox with tteck script

### Device information

Manufacturer: Fibaro
Model name: FGRM 222
Node ID: ...

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [X] The log includes a re-interview of the problematic device _(if applicable)_.

- [X] The log includes the problematic interaction with the device _(if applicable)_.

- [X] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

[zwavejs_2024-07-04.log](https://github.com/user-attachments/files/16097227/zwavejs_2024-07-04.log)
</div>

AlCalzone, 2024-07-09:
@Patator33 please test [this config file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/8f0bfcc81bd3a21cb1e9f589347234e6d17dce02/packages/config/config/devices/0x010f/fgrm222_0_24.23.json). [This page](https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files) describes how to do that.

Note that you have to restart Z-Wave JS UI and re-interview the device afterwards.

Patator33, 2024-07-09:
I have do this and no multilevel switch.
Any other test to do ?

AlCalzone, 2024-07-09:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

The log should include the startup of the driver (this happens when you enable **log to file** in the settings and save), as well as a re-interview.

wigster, 2024-07-09:
I've restarted my ZwaveJS a couple of time after placing the device file into store/config into a file named fgrm222 (not sure this is correct and I do not know how to tell it's being picked up).

Reinterviewing node 23 does has not brought back the multilevel switch. Drvier debug log attached
[zwave-js-ui-store.zip](https://github.com/user-attachments/files/16141964/zwave-js-ui-store.zip)

AlCalzone, 2024-07-09:
> Drvier debug log attached

That's not the driver log.

> a file named fgrm222

It needs to be end in `.json`, otherwise it won't be picked up.

wigster, 2024-07-09:
Ok, I've renamed the file in config. The multilevel switch is still missing after interviews. I've noticed there is a ManufacturerProprietary parameter which allows me to control the blinds including tilt fine. It does not get picked up by HA unfortunately.

Here is the new driver log. Node 23 is the one that I reinterviewed.

[zwave-js-ui-store (1).zip](https://github.com/user-attachments/files/16142982/zwave-js-ui-store.1.zip)

wigster, 2024-07-09:
Yes this works now! HA repair picks up that there is a new config file. Brilliant. Thanks!

Patator33, 2024-07-09:
It's working for me. Sorry for my last message.
Thanks a lot !

Patator33, 2024-07-09:
Juste one thing, the manufacturer, the product and the product code have desappear.
Can we restore these information into zwavejsui by modifying the json file ?

AlCalzone, 2024-07-09:
> switch entity

Do you mean the Binary Switch? I can remove that too.

wigster, 2024-07-09:
Yes I mean exactly the binary switch. No need for it since the multivalue switch handles it all.

AlCalzone, 2024-07-09:
Ok, should be included in one of the upcoming releases. You can then remove your custom file afterwards.