zwave-js/zwave-js #7029: Zooz temp/humidity sensors stop reporting humidity on 9.16.0

sholdee, 2024-07-19:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

9.16.0

### ZwaveJS version

1.37.0

### Describe the bug

My ZSE40 and ZSE44 sensors stopped reporting humidity after [updating to 9.16.0](https://github.com/sholdee/home-ops/commit/c187113afde386a616801730848895370a4229ea) from 9.15.0.

After excluding and re-including the ZSE40 to troubleshoot, the humidity was not even listed as a sensor value in the UI. Only temperature was listed.

[Reverting to 9.15.0 ](https://github.com/sholdee/home-ops/commit/472ac9dc42c99546a09fdd4c94069a7aae8d358e) resolved the issue. You can see when the upgrade and downgrade took place looking at these graphs.

[image]
[image]

### To Reproduce

Upgrade zwave-js-ui to 9.16.0 with a ZSE44 or ZSE40 sensor.

### Expected behavior

Humidity values are reported.

markruys, 2024-07-19:
I noticed for a ZW100 sensor weird behavior since upgrade 9.15 to 9.16. Temperature in HA gets wrongly interpreted (it shows -5 °C instead of 21 °C). In Zwave-ui-js the value of this device is flapping between °C and °F without clear pattern to me. I'm still investigating, and perhaps file a bug report, but it could be related to this issue. I reverted back to 9.15, so far HA  shows correct 21 °C again.

AlCalzone, 2024-07-19:
I can reproduce, fix is coming

markruys, 2024-07-23:
Yes indeed, the issue I faced seems to be gone. Many thanks @AlCalzone 👍