zwave-js/zwave-js #7050: Missing device configuration: homeSYS (GroupSYS) HomeMech-2001 Dimmer

austwhite, 2024-07-28:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

homeSYS (GroupSYS) HomeMech-2001 Dimmer

### Manufacturer ID

816

### Product Type

514

### Product ID

53276

### Firmware Version

1.38

### Do you have a manual?

[HomeMech-2002 Instruction.pdf](https://github.com/user-attachments/files/16402231/HomeMech-2002.Instruction.pdf)

I found a manual for the HomeMech-2002 which is the push button version of the HomeMech-2001.  I believe they are virtually identical, aside from the 2001 having the rotary dimmer and the 2002 having a push button.

I couldn't find the config for the homeSys HomeMech-2001 or 2002 dimmers in the Z-Wave JS listing.   I hope I have posted this correctly.   Not sure if these are rebranded, I couldn't find a listing on the z-wave alliance pages for this particular dimmer.

austwhite, 2024-07-28:
I was doing some research and found this device may also be listed as Shenzhen Sunricher Technology Limited,   Model: SR-ZV2835PAC

austwhite, 2024-08-02:
Updated Z-wave certification link with information provided by homeSYS
https://products.z-wavealliance.org/products/3331?selectedFrequencyId=1

AlCalzone, 2024-10-30:
@zwave-js-bot import config 3331 from zwa