zwave-js/zwave-js #7052: GetFrequencyInfo not supported on Zniffer with FW 2.43

AlCalzone, 2024-07-29:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/7036

<div type='discussions-op-text'>

<sup>Originally posted by **@mundschenk-at** July 21, 2024</sup>
### Checklist

- [X] I have read and followed the above instructions
- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there or the fix **did not work**.

### Describe the issue

What is happening?
I own an official (black) Sigma Design branded zniffer (ACC-UZB-S). Using the official Windows app, I can record wireless traffic but decoding S2 messages fails for me (copying the keys from zwave-js-ui). When trying to use the USB stick with the ZWave-JS zniffer feature, the stick does not work:

```
12:58:22.870 ZNIFFR   received Zniffer info:
                        Chip type:       ZW050x
                        Zniffer version: 2.43
2024-07-21 12:58:23.886 INFO ZNIFFERMANAGER: Retrying in 5s...
2024-07-21 12:58:24.189 DEBUG SOCKET: New connection <CONNECTIONSTRING>
2024-07-21 12:58:24.225 DEBUG SOCKET: Event INITED emitted to < CONNECTIONSTRING >
2024-07-21 12:58:24.233 DEBUG SOCKET: Event ZNIFFER_API emitted to < CONNECTIONSTRING>
2024-07-21 12:58:24.237 INFO APP: Zniffer api call: undefined
2024-07-21 12:58:28.905 ERROR ZNIFFERMANAGER: Zniffer error: Error Resource temporarily unavailable Cannot lock port
```

What did you expect to happen instead?
Being able to zniff traffic.

### Software versions

Driver (node-zwave-js):  13.0.1

Z-Wave JS UI:  9.16.1

If you are using something non-standard, tell us here: Currently trying to run on a standalone RPi 4 using NPM, but I previously also tried things in the HA add-on in a VM.

### Checklist

- [X] I [made sure](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index?id=rule-1-use-the-driver-log-luke) to provide a **driver log** on level **debug**.

- [ ] The log includes a re-interview of the problematic device _(if applicable)_.

- [ ] The log includes the problematic interaction with the device _(if applicable)_.

- [ ] I provided the node ID of the problematic device _(if applicable)_.

### Upload Logfile

No driver log created (because this instance is only running the Zniffer). There's nothing in the `zniffer_current.log` that I haven't included above.</div>

AlCalzone, 2024-07-29:
The problem is that Zniffer FW 2.43 does not support the `GetFrequencyInfo` command, which Z-Wave JS expected to be supported. https://github.com/zwave-js/node-zwave-js/pull/7051 should fix this by assuming the region names are the same as in 2.55

AlCalzone, 2024-07-29:
If you want, you can install a test version in Z-Wave JS UI:
https://github.com/zwave-js/node-zwave-js/pull/7051#issuecomment-2255242159

mundschenk-at, 2024-07-29:
I checked it out, works great!

AlCalzone, 2024-07-30:
fixed by #7051

closed 2024-07-30 by AlCalzone