zwave-js/zwave-js #7053: Missing device configuration: Ness Z-Wave Smart Plug ZA-216001

kizza42, 2024-07-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Ness Z-Wave Smart Plug

### Manufacturer ID

0x0208

### Product Type

0x0102

### Product ID

0x0029

### Firmware Version

1.07

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4196

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/4196/Manual-SmartPlug.pdf

kizza42, 2024-07-29:
You can copy config params from https://devices.zwave-js.io/?jumpTo=0x0208:0x0100:0x0019:0.0

AlCalzone, 2024-10-30:
@zwave-js-bot import config 4196 from zwa

PR #7767 (2025-04-28): feat(config): add Ness Smart Plug ZA-216001
Fixes #7053 

Parameters set up to match HKZW-S008, except for parameter 27 where the manual for this device shows the options reversed from what is in the database for the HKZW-S008.
closed 2025-04-28 by AlCalzone