zwave-js/zwave-js #7086: Unable to create association between motion sensor and outlet/switch

haraldpaulsen, 2024-08-05:
First let me say that this is something that used to work in the distant past, before the road to certification started and one became stricter with allowed associations.

Device A: Fibaro Motion Sensor FGMS001
Device B: Fibaro Metered Wall Plug Switch FGWP102

Both devices are included with default options, ie no security.

Device A, the motion sensor, has an association group 4, called Motion BC, that sends a Basic ON/OFF on motion and motion cleared
Device B, the wall plug, supports Binary Switch v1, which I assume is the Basic ON/OFF.

When trying to create an association I get the error: 
Association not allowed: Node A sends no commands in this group that Node B supports!
(actual numbers replaced by A and B)

I'm not very familiar with the code itself, but it appears that there is a bug in the function that checks for compatible commands.

Z-Wave JS is the newest version as of 2024-08-05, same with Home Assistant.

AlCalzone, 2024-08-05:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

The log must include a re-interview of both devices.

haraldpaulsen, 2024-08-10:
I'll get on it

closed 2024-08-23 by zwave-js-assistant[bot]