zwave-js/zwave-js #7090: Update from 9.16.2 to 9.17.0 resulted in lost of half of the zwave network.

Judman, 2024-08-09:
Hello,

Last night I updated from 9.16.2 to 9.17.0. When the update was complete only 10 of my 20 devices are showing on the network.

I restored the 9.16.2 backup HA made, but that did not fix the problem.

I am using HA version 2024.7.4 with zwavejs and zwavejs-iu as addons. I am using a Zooz 800 Series GPIO Module over Ethernet in a Olimex ESP32-PoE with ESPHome firmware , FW v1.20 SDK v7.19.3.  https://tubeszb.com/product/z-wave-poe-kit/

Attached are the logs I have and node config. I really do not know where to even start troubleshooting. I was thinking I might try to restore a proxmox backup from 4 days ago, but if the issues is with the controller it would be a waste.

[zwavejs_2024-08-08.log](https://github.com/user-attachments/files/16561031/zwavejs_2024-08-08.log)
[nodes.json](https://github.com/user-attachments/files/16561037/nodes.json)
[settings.json](https://github.com/user-attachments/files/16561038/settings.json)
[zwavejs_current.log](https://github.com/user-attachments/files/16561039/zwavejs_current.log)
[zwavejs_current.log](https://github.com/user-attachments/files/16561043/zwavejs_current.log)

[image]
[image]

Judman, 2024-08-09:
@robertsLando I wasn't sure where to post this. I figured since updating the zwaveju-ui seems to cause the issue that was the right place. My Apologies!

robertsLando, 2024-08-09:
@Judman No worries :)

Judman, 2024-08-11:
Here is another log showing a restart before the update with all nodes recognized.

https://pastebin.com/bcKEEbSz

AlCalzone, 2024-08-12:
Looks like the controller "forgot" about the nodes for some reason. No idea how that can even happen, other than cutting power right when the controller was re-writing its memory.

If you have an NVM backup from before it happened, you can try restoring that.
If not, you can make a new one, send it to me and I can take a look if the information is still recoverable.