zwave-js/zwave-js #7096: Take over network from existing controller

AlCalzone, 2024-08-12:
To make migration from legacy controllers easier, especially when NVM backup/restore is not an option, consider the following flow:

- Start learn mode, join existing network. As part of this, existing nodes' info is transferred to Z-Wave JS
- Ask user to turn off existing controller
- Read NVM, modify NVM to set own node ID to the node ID of the previous SUC
- Write modified NVM
- Restart to apply modified NVM
- Maybe (investigate?): Change own role to SUC/SIS

AlCalzone, 2024-08-15:
If your Z-Stick is on firmware 1.2 you can just take an NVM backup and restore that on a new controller using Z-Wave JS UI.

TomMalfrere, 2024-08-15:
> If your Z-Stick is on firmware 1.2 you can just take an NVM backup and restore that on a new controller using Z-Wave JS UI.

  "productLabel": "ZW090",
  "productDescription": "Z‐Stick Gen5 USB Controller",
  "manufacturer": "AEON Labs",
  "firmwareVersion": "1.0",
  "sdkVersion": "6.51.2",
  "protocolVersion": 3,

AlCalzone, 2024-08-15:
It's possible that you can upgrade with the firmware updater from Aeotec (search for Z-Stick 5 firmware 1.2).

If not then this feature would be for you.

TomMalfrere, 2024-08-20:
I tried updating the firmware this weekend. It did not work.
It appears my stick is running since september 2017 so it was likely manufactured before 2018...

Hedda, 2024-10-10:
Sounds promising if can make the whole procesure user friendly all from the UI.

For reference, Home Assistant’s ZHA integation which is their native Zigbee gateway component has a succeded with the user friendly part via similar built-in UI feature for ”radio migration” that allow users to migrate from and old to a new Zigbee Coordinator adapter via a scripted backup and restore function that does everything for the user. See:

https://www.home-assistant.io/integrations/zha#to-migrate-to-a-new-zigbee-coordinator-radio-inside-zha

This feature in the ZHA integration makes it super simple for any user to migrate to a new Zigbee Coordinator (radio adapter).

 https://connectzbt1.home-assistant.io/migrate-zigbee-zha-only/

Would be awesome if the Home Assistant’s Z-Wave JS integation and Z-Wave JS UI had similar features that made it just as easy and user-friendly to migrate to a newer Z-Wave Controller.

Hedda, 2024-10-10:
Oh, and by the way, also for reference, both Nerivec’s ”ember-zli” and ”zigpy-cli” command line interface tools similarly convert Zigbee network backups to similar JSON representation that can be modified before restore backup from file

https://github.com/Nerivec/ember-zli

might be interesting to look at Nerivec’s ”ember-zli” because it is made for Silicon Labs Gecko Bootloader and written in TypeScript.

Perhaps might also be interesting to look at the Open Zigbee coordinator backup format for a consistant JSON representation (even if that was made for Zigbee and not Z-Wave):

https://github.com/zigpy/open-coordinator-backup

Though zigpy-cli and bellows are what is used by ZHA but those are written in Python:

https://github.com/zigpy/zigpy-cli

https://github.com/zigpy/bellows

brandond, 2024-11-25:
I recently had to decommission an old Vera Home controller, and I think all the info necessary to take over would have been available in the backup files - but I ended up just excluding/including all the devices into a new network because I couldn't figure out how to get zwave-js to take over. This would have been great!

Hedda, 2025-01-09:
> * Start learn mode, join existing network. As part of this, existing nodes' info is transferred to Z-Wave JS
> …

@AlCalzone Any updates on the needed parts to make it possible to join a secondary controller to an existing Z-Wave network?