zwave-js/zwave-js #7109: Unable to Remove or Replace Dead Node

mwolter805, 2024-08-14:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Home Assistant Add-on

### Z-Wave JS UI version

9.17.0

### ZwaveJS version

13.1.0

### Describe the bug

Setting up a new system for a family member and had a couple nodes that were intermittently unavailable.  Attempted to do a 'replace' for one of them but could not. The other did the normal exclusion and inclusion and it worked fine. All steps were performed through the zwaveJSui GUI. 

Now the node (7) that we attempted to replace is stuck is stuck in the zwaveJSui database. We have already added it as a new node (16) but the original node 7 is stuck as dead and cannot be removed.

### To Reproduce

Factory default one of the nodes and attempted to replace it. Waited for node status to be "dead" and started the replace in Nodes Manager. Received the following error. Only able to get partial text as the error appears briefly.

```
or while calling replaceFailedNode: The node replace process could not be started due to the following reasons:
· Node 7 is not in the list of failed nodes (ZW0362)
```

Then tried to remove the node through Node -> Advanced -> Failed Nodes and received the following error.
```
Error while calling removeFailedNode: The node removal process could not be started due to the following reasons: · Node 7 is not in the list of failed nodes (ZW0360)
```

Waited several hours and tried to remove the node again and received the same error. 

### Expected behavior

Be able to replace or remove a dead or failed node. 

### Additional context

<details>

<summary>Node Debug</summary>

```json
{
  "id": 7,
  "name": "Greenhouse Heater Dead",
  "loc": "Greenhouse",
  "values": [],
  "groups": [],
  "neighbors": [
    1,
    3,
    5,
    6,
    8,
    9,
    10,
    13,
    14
  ],
  "ready": false,
  "available": false,
  "hassDevices": {},
  "failed": false,
  "inited": false,
  "eventsQueue": [
    {
      "time": "2024-08-14T12:59:00.419Z",
      "event": "dead",
      "args": [
        0
      ]
    },
    {
      "time": "2024-08-14T12:59:00.425Z",
      "event": "interview failed",
      "args": [
        {
          "errorMessage": "The node is dead",
          "isFinal": true
        }
      ]
    }
  ],
  "status": "Dead",
  "interviewStage": "ProtocolInfo",
  "priorityReturnRoute": {},
  "customReturnRoute": {},
  "customSUCReturnRoutes": [],
  "applicationRoute": false,
  "hexId": "0xXXXX 0xXXXX-0xXXXX",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0xXXXX:0xXXXX:0xXXXX:0.0",
  "productLabel": "Unknown product 0xXXXX",
  "productDescription": "0xXXXX",
  "manufacturer": "Unknown manufacturer 0xXXXX",
  "protocolVersion": 3,
  "nodeType": 1,
  "endpointsCount": 0,
  "endpoints": [
    {
      "index": 0,
      "label": "Root Endpoint"
    }
  ],
  "isSecure": false,
  "security": "None",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 16,
    "specific": 1
  },
  "lastActive": 1723586434440,
  "firmwareCapabilities": {
    "firmwareUpgradable": false
  },
  "protocol": 0,
  "deviceId": "undefined-undefined-undefined",
  "_name": "Greenhouse Heater Dead (Greenhouse)",
  "statistics": {
    "commandsTX": 0,
    "commandsRX": 0,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0
  },
  "prioritySUCReturnRoute": false,
  "errorReceive": false,
  "errorTransmit": false
}
```

</details>

[zwavejs_2024-08-13.log](https://github.com/user-attachments/files/16613665/zwavejs_2024-08-13.log)

Question: What are the steps a dead node becomes failed? How long does it take? Is there anyway to manually make a dead node status become failed?

AlCalzone, 2024-08-14:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of you trying to remove the node, **but on loglevel `debug`** and attach it here as a file (drag & drop into the text field).

mwolter805, 2024-08-14:
> Which controller are you using? Which firmware version is running on it?

<details>

<summary>Debug info for the controller</summary>

```json
{
  "id": 1,
  "name": "Smartstick",
  "loc": "",
  "values": [],
  "groups": [],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "eventsQueue": [
    {
      "time": "2024-08-14T20:34:23.132Z",
      "event": "alive",
      "args": [
        0
      ]
    },
    {
      "time": "2024-08-14T20:34:23.133Z",
      "event": "ready",
      "args": []
    },
    {
      "time": "2024-08-14T20:34:23.133Z",
      "event": "ready",
      "args": []
    }
  ],
  "status": "Alive",
  "interviewStage": "Complete",
  "priorityReturnRoute": {},
  "customReturnRoute": {},
  "customSUCReturnRoutes": [],
  "applicationRoute": false,
  "hexId": "0x027a 0x0004-0x0610",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x027a:0x0004:0x0610:1.40",
  "manufacturerId": 634,
  "productId": 1552,
  "productType": 4,
  "deviceConfig": {
    "filename": "/data/db/devices/0x027a/zst39lr.json",
    "isEmbedded": true,
    "manufacturer": "Zooz",
    "manufacturerId": 634,
    "label": "ZST39 LR",
    "description": "800 Series Long Range USB Controller",
    "devices": [
      {
        "productType": 4,
        "productId": 1552
      }
    ],
    "firmwareVersion": {
      "min": "0.0",
      "max": "255.255"
    },
    "preferred": false,
    "metadata": {
      "comments": []
    }
  },
  "productLabel": "ZST39 LR",
  "productDescription": "800 Series Long Range USB Controller",
  "manufacturer": "Zooz",
  "firmwareVersion": "1.40",
  "sdkVersion": "7.22.0",
  "protocolVersion": 3,
  "endpointsCount": 0,
  "endpoints": [
    {
      "index": 0,
      "label": "Root Endpoint"
    }
  ],
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": true,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 2,
    "generic": 2,
    "specific": 1
  },
  "lastActive": 1723667663910,
  "firmwareCapabilities": {
    "firmwareUpgradable": false
  },
  "protocol": 0,
  "deviceId": "634-1552-4",
  "hasDeviceConfigChanged": false,
  "rfRegions": [
    {
      "value": 2,
      "text": "Australia/New Zealand"
    },
    {
      "value": 8,
      "text": "China"
    },
    {
      "value": 255,
      "text": "Default (EU)"
    },
    {
      "value": 0,
      "text": "Europe"
    },
    {
      "value": 3,
      "text": "Hong Kong"
    },
    {
      "value": 5,
      "text": "India"
    },
    {
      "value": 6,
      "text": "Israel"
    },
    {
      "value": 32,
      "text": "Japan"
    },
    {
      "value": 33,
      "text": "Korea"
    },
    {
      "value": 7,
      "text": "Russia"
    },
    {
      "value": 9,
      "text": "USA (Long Range)"
    }
  ],
  "supportsLongRange": true,
  "supportsTime": false,
  "_name": "Smartstick",
  "statistics": {
    "messagesTX": 21,
    "messagesRX": 40,
    "messagesDroppedRX": 0,
    "NAK": 0,
    "CAN": 0,
    "timeoutACK": 0,
    "timeoutResponse": 0,
    "timeoutCallback": 0,
    "messagesDroppedTX": 0
  },
  "prioritySUCReturnRoute": false,
  "lastReceive": 1723667663910,
  "lastTransmit": 1723667663455,
  "errorReceive": false,
  "errorTransmit": false,
  "powerlevel": 6,
  "measured0dBm": 0,
  "RFRegion": 9
}
```

</details>

RF Region is set to USA (Long Range) and everything appears to work well except not being able to remove this stuck node.

AlCalzone, 2024-08-15:
Please make another [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of you trying to remove that device, **on loglevel `debug`** and attach it here as a file (drag & drop into the text field).

Please also make an NVM backup (to be found in the wand menu) and share that too.

mwolter805, 2024-08-15:
@AlCalzone Attached are the requested files

12:45:05 log started
12:46:50 remove started
12:49:40 received remove failed notification in GUI

[zwavejs_current.txt](https://github.com/user-attachments/files/16625385/zwavejs_current.txt)
[NVM_20240815125416.bin.zip](https://github.com/user-attachments/files/16625392/NVM_20240815125416.bin.zip)

AlCalzone, 2024-08-15:
Here's an archive containing the fixed NVM: [NVM_fixed.zip](https://github.com/user-attachments/files/16630594/NVM_fixed.zip).
Just restore it and it should be back to normal. Some routes may not be correct, but at least they shouldn't lock up the controller.

Disclaimer: While this functionality is well-tested, there's always a risk of something going wrong and possibly bricking the controller. I don't take responsibility if that happens.

mwolter805, 2024-08-16:
That worked. Thank you. What's the best way to send a tip?

AlCalzone, 2024-08-16:
There are a few options on the project page. Github Sponsors or PayPal are probably the best.

closed 2024-08-16 by AlCalzone