zwave-js/zwave-js #7123: Add Zooz ZSE70 Outdoor Motion Sensor

jhmartin, 2024-08-25:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

- [X] The device was interviewed completely and has no missing or `undefined` IDs

### Which device is missing?

Zooz ZSE70 Outdoor Motion Sensor

### Manufacturer ID

0x027a

### Product Type

0x0004

### Product ID

0x0006

### Firmware Version

1.20

### Do you have a manual?

https://www.support.getzooz.com/kb/article/1726-zse70-outdoor-motion-sensor-specs/

jhmartin, 2024-08-25:
This is probably in the ZWave catalog, but the site is misbehaving for me: "There are over 0 interoperable Z-Wave products throughout the world" and returns no search results.

jhmartin, 2024-08-25:
[zooz.txt](https://github.com/user-attachments/files/16742303/zooz.txt)
Debug info attached.

NellicusLabs, 2024-09-28:
+1. I was having an issue and then found this issue for the sensor not currently being present in zwave-js on HomeAssistant

MichaelStruck, 2024-10-26:
Just purchased this and noticed it is not listed in the database or in Zwave JS (device: 0x027a 0x0004-0x0006). Operational, however, in Home Assistant as is.

hceuterpe, 2024-11-08:
The device was added to the Z-Wave Alliance DB 4 days ago.  I'd be interested in helping to add this (I own four of these).  I haven't done this before though so I'll have to spend some time to get up to speed on how to generate.

AlCalzone, 2024-11-08:
@zwave-js-bot import config 5150 from zwa

AlCalzone, 2024-11-08:
Let's hope I didn't break that bot command recently 🤞🏻

AlCalzone, 2024-11-08:
@hceuterpe if you are going to do this, I really can't stress enough editing the files in VSCode, like the documentation states. We have an extension for that, which allows you to see a preview of the parameter definitions and which highlights common errors:
[image]

hceuterpe, 2024-11-08:
I finished up the revisions.  That modified file passed the config linting command without no errors or warnings.  It really wasn't that difficult.  I always wondered if there was a way to convert the Z-wave DB XML files into the json config files.  Glad to know where it is now :) 

I'm testing it out now in my system.  If someone else here has a devices and they want to test it out, let me know and I can attach it so you can try out.

My only question is this:  The majority of the 800LR devices Zooz has released do not distinguish the 800LR in either the device label or filename (actually it seems all but one, ZEN37).  ZSE70 also hasn't existed before the 800LR chipset revision.  Do we want to just shorten the filename and label to ZSE70?

closed 2025-01-10 by AlCalzone